riscvarchive/riscv-v-spec #185: Example(6.4) in specification document

eroom1966, 2019-05-22:
We are in the process of testing all of the code fragment examples in the specification document, against the Imperas simulation model, we have a question regarding the example in section 6.4

For the first example, there is no function prototype defined
could you please 
1. describe the expected function arguments
2. specifically state what values are held in
- x10 (note x10 is the equiv of ABI a0, is this a mistake?)

Many Thx
Lee Moore - Imperas

jim-wilson, 2019-05-22:
It is just a code sample to show how to write a useful loop.  It isn't meant to be an entire function.

a0 holds the number of elements in the vectors.  a1 is the address of the input vector, which is a pointer to 16-bit elements.  a2 is the address of the output vector, which is a pointer to 32-bit elements.

Yes, the use of x10 looks like a bug.  It would make more sense to use a3 instead.  This would then be the value we are multiplying by.

eroom1966, 2019-05-24:
Hi Jim
Thanks for this, I have this working now
at some point do you want me to send the (working) code back to put in as your examples ?
Thx
Lee

closed 2019-05-24 by eroom1966

jim-wilson, 2019-05-25:
You are welcome to submit a pull request to suggest changes, or file another issue to report the bug.