riscvarchive/riscv-v-spec #190: Should vector mask logical instructions really give -1, -2 results?

jhauser-us, 2019-05-25:
As currently defined, the complementing vector mask logical instructions VMNAND, VMNOR, and VMXNOR will turn ordinary "true/false" operands with values 0 or 1 into results with values &minus;1 or &minus;2, which at various times will be both surprising and inconvenient.  (I'm using two's complement notation here merely to be succinct.  The important point is that the results are outside the expected 0-or-1 range.)

In the discussion for an earlier issue (#139), it was said that "software should not particularly care".  That will not always be the case, and in any event it's certain that programmers will sometimes be forced to notice that &minus;1 means *true* and &minus;2 means *false* in addition to the usual 0 and 1.

As an instructive case, note that in section 12.7, "Vector Integer Comparison Instructions", the proposed pseudoinstruction

    vsge{u}.vx vd, va, x

will not act the same as the hardware's comparison instructions.  While the hardware's comparisons fill the destination vector elements with values of 0 or 1, these VSGE and VSGEU pseudoinstructions will fill the elements with values &minus;1 or &minus;2, because they use VMNAND.  In my view, this causes the pseudoinstructions to fail if implemented as proposed in the document.  To get the expected results, a different instruction sequence must be used, but I personally was unable to find one that didn't require an additional vector register as a temporary.

It is not true that the difference between &minus;2 and 0, or between &minus;1 and 1, will never matter to software.  When LMUL = 1, it is possible to use VSLT alone as a vector implementation of the C less-than operator `<`, but it is not possible to use the proposed VSGE pseudoinstruction alone as a vector implementation of the C greater-than-or-equal operater `>=`, because the result elements are required to be 0 or 1.  It may be believed that such cases are not of sufficient importance to influence the hardware, but they should not be said to be nonexistent.

At the very least, readers deserve a warning about this behavior.

jhauser-us, 2019-05-27:
My suggestion would be to require that all vector mask-register logical instructions zero all bits of the destination mask register except for bit 0 of each element.  This would be consistent with most (all?) other instructions that write to a mask register destination, including the vector comparison instructions and VMSBF.M, VMSIF.M, and VMSOF.M.

kasanovic, 2019-06-01:
Task Group agreed to change to boolean results for mask logical operations.  The rationale is that software would sometimes prefer the boolean form, the existing bitwise logical instructions can be used when that is desired, and boolean result logic is already there for other mask results.

kasanovic, 2019-06-01:
With this new definition, the vmmv.m pseudoinstruction would not match the other uses of mv in architecture where bits are transported unchanged.  The vmmv.m pseudoinstruction is therefore changed to a vmcpy.m (vector mask copy) instruction.

commit 1924988f (2019-06-01): Changed behavior of mask-logical instructions to write boolean values (zero-extended 0 or 1 value) instead of performing bitwise logical operations.
Also changed name of pseudo-instruction vmmv.m to vmcpy.m to signify that it no longer does a bitwise copy.
Closes #190.
closed 2019-06-01 by kasanovic