riscvarchive/riscv-v-spec #192: Vector Load/Store Segment Instructions with LMUL>1.

JerryShih, 2019-05-27:
In [spec](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#78-vector-loadstore-segment-instructions-zvlsseg), the LMUL should be 1 for vector load/store segment instructions.
Why we can't use register group for these instructions?

Here is the example c code:
```
void cmplx_conj_f32(float *pSrc, float *pDst, unsigned int num)
{
  while (num > 0u) {
    *pDst++ = *pSrc++;
    *pDst++ = -*pSrc++;
    num--;
  }
}
```
If we only use LMUL=1, the vector code could be:
```
    vsetvli iVL, iNum, e32, m8
    addi pSrcImag, pSrc, 4
    addi pDstImag, pDst, 4
    # use 2 vlsw instructions with stride 8
    vlsw.v vReal, (pSrc), iStride
    vlsw.v vImaginary, (pSrcImag), iStride
    vfsgnjn.vv vImaginary, vImaginary, vImaginary
    vssw.v vReal, (pDst), iStride
    vssw.v vImaginary, (pDstImag), iStride
    .........
```
If we could set LMUL with vector load/store segment instructions, the could could be:
```
    vsetvli iVL, iNum, e32, m8
    # load data to vReal and vImaginary
    vlseg2w.v vReal, (pSrc)
    vfsgnjn.vv vImaginary, vImaginary, vImaginary
    vsseg2w.v vReal, (pDst)
    ....
```

aswaterman, 2019-05-27:
In this case, it would be more efficient on most machines to use a
unit-stride load, followed by a masked negation that only negates every
second element, followed by a unit-stride store. No need for segments.

solomatnikov, 2019-05-27:
I agree with @JerryShih on this - `LMUL` requirement should be relaxed for segment loads and stores.

Segment loads and stores are also used in fast FFT implementations (as well as in other pieces of SW) and at least some versions and phases of these implementations do not need many vector registers. Allowing `LMUL > 1` would help to improve performance and/or reduce the number of physical bits in vector register file. This would help vector implementations to achieve competitive PPA.

JerryShih, 2019-05-28:
> In this case, it would be more efficient on most machines to use a unit-stride load, followed by a masked negation that only negates every second element, followed by a unit-stride store. No need for segments.

@aswaterman 
Yes, we could read the real part and imag part with one load(I think vlw in this case) and use a mask with vfsgnjn for the imag part. But it's not very easy to generate the mask. It's easy to separate the real part and imag part just using vlseg2w. If we could have LMUL>1, we could utilize more vector registers.

aswaterman, 2019-05-28:
You can generate the mask by setting SEW = (2 * (real element width) / (desired LMUL)), and setting LMUL = 1, then splatting a constant '1' to v0 using VMERGE.

JerryShih, 2019-05-28:
@aswaterman 
Thank you for the mask tutorial.

But how about the following c code?
I need to setup various masks if I try to load the floats as much as possible. 
```
float32_t real_sum = 0.0f, imag_sum = 0.0f;
float32_t a0, b0, c0, d0;
while (numSamples > 0u) {
  a0 = *pSrcA++;
  b0 = *pSrcA++;
  c0 = *pSrcB++;
  d0 = *pSrcB++;

  real_sum += a0 * c0;
  imag_sum += a0 * d0;
  real_sum -= b0 * d0;
  imag_sum += b0 * c0;

  /* Decrement the loop counter */
  numSamples--;
}
```
I will use 4 vlsw instructions to load pSrcA and pSrcB into 4 vector register groups. Then, it's easy to do the remaining computing. Is this implementation slower than the mask-base implementation?

aswaterman, 2019-05-28:
If using unit-strided loads and stores is an option, it's almost always going to perform better than the alternatives, even if there's a bit of extra setup code.

There have been discussions about adding a complex-multiply instruction using the EDIV mechanism, which would be a perfect match to this code.  It would allow the use of unit-strided loads and stores, without any extra effort to generate masks.  Even without that new support, I think that unit-stride loads and stores are likely to perform better for this code on most machines.

Supporting segmented loads with larger LMUL is a discussion separate from these specific code examples, and one that @kasanovic should provide his feedback on.

kasanovic, 2019-06-01:
An 8-segment load with LMUL=4 would touch all 32 registers.

Task group decision was to allow LMUL>1 for segment load/stores, but to constrain the product of LMUL and nf to be 8 or less.  This number is how many vector registers are touched by the segment load or store (equal to any plain LMUL operation), with each segment load/store constrained to moving 1/4 of total vregfile size.

closed 2019-06-01 by kasanovic

JerryShih, 2019-06-03:
@kasanovic 
If we use vlseg2w v0,(pSrc) with lmul=4, will the first 2 word data saved in v0 and v4?

I my thought, the v0 to v3 are in the same register group. So, we treat v0-v3 and v4-v7 as the two big vector registers.

kasanovic, 2019-06-03:
Yes, the individual results must be the same as any other vector register group.

commit 5d82fd51 (2019-06-01): Specified that segment load/stores can have LMUL>1.
Closes #192.