riscvarchive/riscv-v-spec #193: Dose floating-pointing compute instruction check NaN when FLEN > SEW ?

chihminchao, 2019-05-30:
If No,
     1. it is possible to load double-precision by fld, compute it as single point in  vector and store back to floating-point register.  The result is meaningless
     2.  implementation such as spike needs two different API to access the floating-point register

Based on v-spec 11.1
"For floating-point operation, the scalar can be taken from a scalar f register. If FLEN>SEW, the least-significant bits of the `f`register are used. If FLEN<SEW, the value is NaN-boxed (one-extended) to SEW."

Based on user-spec  2.2 p56
"Floating-point compute and sign-injection operations calculate results based on the FLEN-bit values held in the f registers. A narrow n-bit operation, where n < FLEN, checks that input operands are correctly NaN-boxed, "

aswaterman, 2019-05-30:
It looks like the spec currently says the answer is “no”, but I’m not sure why. Checking for a NaN box in an f-register when FLEN > SEW seems like a reasonable thing to do. @kasanovic any comment?

kasanovic, 2019-06-01:
This was an oversight - the scalar args to vector-scalar operations should behave the same as regular scalar args.

commit 9903776b (2019-06-01): Floating-point scalar values should be checked for NaN-boxing when FLEN>SEW.
Closes #193.
closed 2019-06-01 by kasanovic