riscvarchive/riscv-v-spec #199: [Minor spec Issue] Using x0 as rd in vsetvli/vsetvl

billhuffman, 2019-06-01:
Section 6.1 of the spec explicitly states what happens if rs1 is x0 for vsetvli/vsetvl but not what happens if rd is x0.  I presume it does everything it would have done but doesn't write an x-register.  The way it's used in examples, however, it might not set vl either.  An extra sentence would avoid ambiguity.

aswaterman, 2019-06-01:
Your presumption's right.  The base ISA spec follows the discipline that using x0 is not special unless explicitly specified.  So, my concern is that explicitly specifying that x0 is _not_ special when used as rd could lead to additional (but unrelated) confusion.

Perhaps we could add a non-normative note.  I'll make a PR and let @kasanovic decide.

billhuffman, 2019-06-02:
Yes, I see what you mean.  The note might simply say that even when rd is x0, vl is still written.

It was vl that was my uncertainty.  From a usage point of view, the most common use of rd as x0 is probably when re-adjusting SEW and LMUL while leaving vl the same.  I presume there it's x0 to avoid additional dependencies.  For that use, it would also be correct not to write vl either.   That's where I'm coming from in thinking a clarification might be helpful.

      Bill

On 6/1/19 3:21 PM, Andrew Waterman wrote:

Your presumption's right. The base ISA spec follows the discipline that using x0 is not special unless explicitly specified. So, my concern is that explicitly specifying that x0 is not special when used as rd could lead to additional (but unrelated) confusion.

Perhaps we could add a non-normative note. I'll make a PR and let @kasanovic<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_kasanovic&d=DwMCaQ&c=aUq983L2pue2FqKFoP6PGHMJQyoJ7kl3s3GZ-_haXqY&r=AYJ4kbebphYpRw2lYDUDCk5w5Qa3-DR3bQnFjLVmM80&m=X_c5ZTtPqTjVfRaBr7-jrNyLIT4MqO8lw0wQeP4YwMQ&s=iqMdc6sYhDDjdK4N2HeV9DozpZBPUyROD8Y7pju1tPU&e=> decide.

closed 2019-06-03 by kasanovic

aswaterman, 2019-06-03:
Added a note about this.

billhuffman, 2019-06-03:
Thanks.      Bill

On 6/3/19 9:27 AM, Andrew Waterman wrote:
EXTERNAL MAIL

Added a note about this.

commit 9f493d92 (2019-06-01): Add note about vsetvl with rd=x0; explain why it is useful

Resolves #199
PR #200 (2019-06-03): Add note about vsetvl with rd=x0; explain why it is useful
Resolves #199