riscvarchive/riscv-v-spec #202: Unclear when thread can be migrated to hart with different VLEN/SLEN/ELEN

jnk0le, 2019-06-04:
Without already mentioned instructions or hints to disable vector state I can assume that `vl=0` is the only "safe" state to be rescheduled to architecturally different core.

But, most examples in appendix A, contains the following control loop:
```
loop:
   vsetvli t0, a2, e8,m8  # Vectors of 8b
   sub a2, a2, t0              # Decrement count
   [...]
   bnez a2, loop               # Any more?
   ret
```
which leave non zero `vl` on function return and potentially permanently locking thread to a "big" or "small" cores.

aswaterman, 2019-06-04:
Not even vl=0 is sufficient, because the regfile still keeps its state in that situation. To effectively support migration, we would need a vdisable instruction to indicate the state is no longer needed. We would also need a significant ABI restriction: applications must not memoize anything that depends on VLEN or SLEN across vdisables.

Making the vector units architecturally identical across cores that threads can migrate to might not be so bad. The small cores can still use much smaller vector units (fewer lanes, more temporal execution); they just need matching regfile parameters.

kasanovic, 2019-06-06:
I think migration between vector units with different VLEN/SLEN should be a non-goal of the spec.   It could be required/implemented by some platform ABI, but the benefits of allowing software to cache these parameters are too valuable (e.g., dynamic linking of libraries tuned for the VLEN/SLEN, software vector-length speculation, precomputing tables tied to VLEN/SLEN, ...)

jnk0le, 2019-06-20:
> Making the vector units architecturally identical across cores that threads can migrate to might not be so bad. The small cores can still use much smaller vector units (fewer lanes, more temporal execution); they just need matching regfile parameters.

In case of restricting it to identical VLENs, we are going to see a bad tradeoffs between regfile capacity in bigger cores and size/power efficiency of lesser ones. SLEN/ELEN are less of an issue.

aka-sps, 2019-07-01:
@kasanovic
How can an RVV program depend on the value of SLEN and how can SLEN be architecturally visible?

jnk0le, 2019-07-02:
> @kasanovic
> How can an RVV program depend on the value of SLEN and how can SLEN be architecturally visible?

Every case of `lmul`>1 due to the way of element packing in grouped registers. Maybe there are more fine cases I'm not aware of.

aswaterman, 2019-07-02:
It's possible the two of you are discussing different cases.  Application code won't expose differences in SLEN if it follows some code-generation rules, primarily keeping the ratio of SEW to VLMUL constant throughout the strip-mine loop.  But these differences will manifest when migrating processes between vector machines with different parameters, even if those code-generation rules are followed.

kasanovic, 2019-12-07:
I'm going to close this issue, as there is no practical proposal beyond requiring all cores involved in migration have same parameters.

closed 2019-12-07 by kasanovic