riscvarchive/riscv-v-spec #205: should vtype vill  bit still read 1 for invalid values in minimum configuration.

David-Horner, 2019-06-06:
In section 3.2. Vector type register: 
"The illegal value represented by `vill` can be encoded
using the illegal 64-bit combination in `vsew[1:0]` without requiring
an additional storage bit."
Must bit XLEN-1 still read 1 for such​? i.e. vill = vsew[1]& vsew[0], so only two storage bits but msb be derived from them.

And because the statement in 3.2.3. Vector Type Illegal vill
"When the vill bit is set, the other XLEN-1 bits in vtype shall be zero."
Would require vsew[1] and vsew[0} be masked to zeros on CSR read when both are set internally.

Is this zero requirement that all other bits be zero too restrictive for the minimal case?

aswaterman, 2019-06-06:
Re: "Must bit XLEN-1 still read 1 for such?" -- I think the definition of `vill` is sufficient to imply the answer is "yes."

Re: "Is this zero requirement [...] too restrictive?" -- I would be OK with eliminating this requirement, and instead saying that when (and only when) `vill` is set, the other fields are permitted to assume invalid values.  Under this alternative, it would be OK for `vsew` to read as `11` when `vill` is high in the simple 32-bit case.

kasanovic, 2019-06-06:
We've tried to avoid implementation-defined behavior in unprivileged spec where possible.  While this one seems fairly innocuous, I can see verification people upset at having to deal with model differences here.

jnk0le, 2019-07-26:
Without taking traps on `vsetvl`, "Zero all" as well as "let it be undefined" rules, would make it impossible to emulate future or unsupported `vtype` configurations.

kasanovic, 2019-09-03:
The current spec uses illegal bit primarily to avoid having to trap CSR writes dependent on data value written, and to allow the vtype register to only support legal values for the current implementation.  It has the other benefit of allowing unprivileged software to determine what vector types are supported but I think this is a minor benefit.

An alternative approach would be trap any illegal write to vtype for emulation purposes (and possibly debugging), but that would then prevent use of this mechanism for user configuration discovery and would require data-dependent traps for this instruction.

A privileged architecture extension could add feature to trap on illegal CSR writes to support this emulation without requiring that all implementations support traps on CSR data values.

jnk0le, 2019-09-03:
I think that the current trapping scheme with an exact `vtype` register storage is the most suitable approach for emulation. (We can also move the "exact `vtype`" requirement to OS profile as to let embedded targets save some silicon)

kasanovic, 2019-11-24:
Closing this issue for now, as we seem to have agreed that future version of privileged architecture can add facility to trap on bad type values for emulation purposes, without adding new awkward data-dependent trap for all implementations.

closed 2019-11-24 by kasanovic

ubc-guy, 2019-11-25:
Sorry to reply to an closed issue, but I wonder if there's been justification for zeroing the entire vtype when vill=1 ?

Leaving the other bits in vtype unchanged seems easier to implement and more useful -- upon requesting an invalid vtype, the application can more easily restore the previous legal version by reading vtype, clearing the MSB, and running vsetvl. If vtype is cleared upon an illegal setting, then the application must always save the value vtype before any vsetvl instruction.

Actually clearing vill could be made even easier. The first way is to force vsetvl to always clear vill upon writing (ie, ignore setting of vill in rs2). This would eliminate the "clearing the MSB" step. The second way would be to allow vtype CSR to be writable, and use the CSRRC instruction to immediately clear vill atomically with reading vtype. There are other hazards of making vtype writable, which I will just leave open for consideration.