riscvarchive/riscv-v-spec #206: Impact of mask value source register v0 on register renaming

dennisyyz, 2019-06-06:
For the effect of vector masking on microarchitecture with register renaming, the spec makes the following explanation:

> However, the arithmetic instructions are designed such that the largest read-port requirement is for fused multiply-add instructions that are destructive and overwrite one source, and hence do not need an extra read port to preserve inactive elements. Given that linear algebra is one of the more important applications for vector units, and that fused multiply-add is the dominant operation in linear algebra routines, microarchitectures will be optimized for fused multiply-add operations and so should be able to preserve inactive elements on other arithmetic operations without large additional cost.

However, there is no other restriction on the reading and writing of the mask value source register v0 itself except for the following description:

> The destination vector register group for a masked vector instruction can only overlap the source mask register (v0) when LMUL=1.

For microarchitecture without register renaming, you can simply read v0 directly. But for microarchitecture with register renaming, you need extra read port to read v0. This means that for a 2-operand instruction with a mask operation, it may require up to 4 read ports: destination register, 2 source registers, and renamed v0.

How does the spec consider and explain this impact?

kasanovic, 2019-06-06:
I don't understand the impact you're referring to.

Yes, the mask register value needs a read port in a renamed vector register file.
The first highlighted text explains that the arithmetic read port count is not greater than that required for fused multiply-adds in a design that zeros masked-off elements.  (four read ports: mask + 3 sources).  This is the same port requirement for non-destructive binary operations with masked-off elements preserved, as you describe.  The point is that preserving masked-off values is not a great additional expense with destructive muladds (though there is a real additional read port cost for vector load instructions).

jovireal, 2019-06-12:
I think the expense might be great, especially when the physical register file is large, and it might not be worthy as most instructions dont need 4 sources.

Instead, I wonder why not consider adding an extra mask register or several instead of reusing v0. Reusing v0 means it cant be used as a normal source or dest. For example, when LMUL is set to 8, only v8 v16 and v24 available, so it cant do widening MAC with masking as there is not enough vreg for use. Though this is an extreme case, I think users might want less restriction in vreg usage.

kasanovic, 2019-06-28:
Longer 64-bit instruction encoding will allow any vector register to be used as mask.
Having separate implicit mask register will only save one vector register (group) and will require additional instructions to load/store or move the mask register to the regular vector registers.
Making mask registers be separate will add considerable area to renamed physical register files (another pool of renames, needing to hold arch version of each register).

kasanovic, 2019-12-07:
Closing this issue as design has stabilized around shared mask/data registers.

closed 2019-12-07 by kasanovic