riscvarchive/riscv-v-spec #210: when SELEN=8, is dot product supported?

jovireal, 2019-06-12:
In zvediv extension, SELEN is introduced as a new implementation parameter. Does SELEN=8 mean no zvediv extension so that no dot product instruction is supported? Or SELEN =8 can support the 8 bit dot production function. As it is also part of the extension, i am confused

kasanovic, 2019-06-28:
SELEN is not defined except when EDIV is supported.
When EDIV is supported, SELEN=8 is maximum value allowed.
E.g., system with ELEN=32 and SELEN=8 supports dot-products as shown in last example of section  19.3

closed 2019-06-28 by kasanovic

lyuchuny3, 2020-03-16:
> SELEN is not defined except when EDIV is supported.
> When EDIV is supported, SELEN=8 is maximum value allowed.
> E.g., system with ELEN=32 and SELEN=8 supports dot-products as shown in last example of section 19.3

Hi, I try to write the assembly for function of example 19.3, but it seems "Illegal instruction", here is my codes:
```      
        # func(int n, int8* a, int8* b , int* c)
        # c[0]=a[0]*b[0]+a[1]*b[1]+a[2]*b[2]+a[3]*b[3] 
        vsetvli   t0, a0, e8
        vlb.v     v0,(a1)
        vlb.v     v1,(a2)
        vsetvli   t0, a0, e32,m1,d4
        vlw.v     v2,(a3)
        vdot.vv   v2,v0,v1
        vsw.v     v2,(a3)
```
could you provide an example for this case?
```
  # Dot product, SEW=32, EDIV=4
  vdot.vv vd, vs2, vs1, vm # vd[i][31:0] += vs2[i][31:24] * vs1[i][31:24]
                                            + vs2[i][23:16] * vs1[i][23:16]
                                            + vs2[i][15:8] * vs1[i][15:8]
                                            + vs2[i][7:0] * vs1[i][7:0]
```
 like example in forms as assembly codes `https://github.com/riscv/riscv-v-spec/blob/master/example/vvaddint32.s`

Thanks

jim-wilson, 2020-03-16:
What exact error are you seeing?  Did you remember to add -march=rv64gcv to enable the vector extension?  With the latest sources you need -march=rv64gcv_zvediv to enable the vector extension and the zvediv extension.