riscvarchive/riscv-v-spec #212: Inefficient use of code space when grouping vector registers

aka-sps, 2019-06-14:
> 3.2.2. Vector Register Grouping (vlmul)

>When vlmul=01, then vector operations on register v n also operate on vector register v n+1, giving twice the vector length in bits. Instructions specifying a vector operand with an odd-numbered vector register will raise an illegal instruction exception.

> Similarly, when vlmul=10, vector instructions operate on four vector registers at a time, and instructions specifying vector operands using vector register numbers that are not multiples of four will raise an illegal instruction exception. When vlmul=11, operations operate on eight vector registers at a time, and instructions specifying vector operands using register numbers that are not multiples of eight will raise an illegal instruction exception.

Consequently, the value of `vlmul` multiplied by the number of vector operands of the instruction determines the number of instruction code bits, which will always be zero.

kasanovic, 2019-07-02:
Not clear what the proposal is. Expand the description.

aka-sps, 2019-07-02:
This is one of a group of issues related to a group of issues regarding the inefficient use of instruction code space and low code density.

There are no specific proposals, but only a remark that the specified bits can be used to encode additional instruction modifications.

For example:
Let we need to convert byte operation to uint64_t result, or float16_t to double precision w/o sequential "widening" (from byte to uint16_t, from uint16_t to uint32_t, from uint32_t to uint64_t or float16_t to float, float to double).
To encode "widening" multiplier we can use lower bits of destination register.

Once again: this is just a reminder for the future (backlog) that a set of operations that will definitely use `vmul > 1` can be encoded without using additional code space.

I understand that you will not change anything in the current version

kasanovic, 2019-12-07:
Point taken, though this does tend to complicate decode as reg specifiers become less regular.

jnk0le, 2020-01-24:
Alternatively it can be encoded in current w2/n2 instructions, as to avoid `lmul>1` limit.

aswaterman, 2020-03-28:
We could consider using this encoding space for quad-widening instructions, at the very slight cost of additional illegal-instruction exception detection complexity.

kasanovic, 2020-03-28:
Adding fractional LMUL proposals make this not workable anymore, as whole rationale for fractional LMUL was to gain access to all register names independent of element widths in use.

Closing this issue.

closed 2020-03-28 by kasanovic