riscvarchive/riscv-v-spec #214: register overlapping rule differs between vslide1up and vslide1down

chihminchao, 2019-06-17:
By the spec [17.4.3](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#1743-vector-slide1up)
"The vslide1up instruction requires that the destination vector register group does not overlap the source vector register group and the mask register if masked. Otherwise, an illegal instruction exception is raised."

By the spec [17.4.4](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#1744-vector-slide1down-instruction)
"The vslide1down instruction requires that the destination vector register group does not overlap the mask register if masked. Otherwise, an illegal instruction exception is raised."

It seems reasonable to use more relaxed rule, used in vslide1down.

aswaterman, 2019-06-17:
As the description for slideup says, this constraint avoids intra-instruction WAR hazards.  It makes sense to apply the same constraint for both slideup and slide1up so they can be implemented with the same hardware.  This constraint also avoids a constraint on restartability: with the weaker constraint, it would be impossible to execute the instruction with nonzero vstart, because part of the input would have been destroyed.

davetw, 2019-06-17:
I felt a little bit confused...{:-))
1) both vslide1up and vslide1down do not support source and destination vreg overlapping
2) both vslide1up and vslide1down support source and destination vreg overlapping
3) vslide1up doesn't allow source and destination vreg overlapping but vslide1down allows overlapping.
spec looks like 3), but Waterman's comment looks like 1)

Meanwhile, about the restartability. Does V extension force the vop to start from index 0? We're wondering to know is it possible to vslide1up from vl-1 to 0 to avoid to destroy the next input data. And to make it possible to support source and destination overlapping.

aswaterman, 2019-06-17:
It's option 3.  In my response I didn't say anything about vslide1down.  I only mentioned vslideup and vslide1up.

The spec allows vslide1up and vslide1down to be restartable (i.e. their behavior is well defined for vstart > 0).

davetw, 2019-06-17:
got it. Now we clear about vslide1up.
Back to vslide1down. 
In section 5.3
> The destination vector register group for a masked vector instruction can only overlap the source mask register (v0) when LMUL=1. Otherwise, an illegal vector instruction exception is raised.

In section 17.4.2
> The destination vector register group for vslidedown cannot overlap the vector mask register if the instruction is masked, otherwise an illegal instruction exception is raised.

these 2 statements look similar. We want to make sure does vslide1down's LMUL limitation?

aswaterman, 2019-06-17:
I think vslidedown’s constraint could be weakened to allow overlap when LMUL=1. @kasanovic any reason for the stricter constraint than usual?

commit 01a1adc5 (2019-11-24): Clarified source/destination overlap rules on slide instructions.
Closes #214.
closed 2019-11-24 by kasanovic