riscvarchive/riscv-v-spec #215: Should we add retriction about dest/src register overlap for Integer/float Comparison Instructions ?

wwang0, 2019-06-17:
As the title, for example LMUL=2:
vmseq.vv v3, v2, vs4
It has WAR hazards without vector register renaming, and we should raise an illegal instruction exception for such case.

aswaterman, 2019-06-17:
The answer should be the same for `vmadc` and `vmseq`.  The text for `vmadc` says, "an illegal instruction exception is raised if the destination vector register overlaps a source vector register group."  (This is stricter than necessary, as there's no WAR hazard when LMUL=1.)

I think we could relax the constraint on `vmadc` to exclude LMUL=1, and apply the same rule to `vmseq`.

wwang0, 2019-06-17:
Agreed.

aswaterman, 2019-06-17:
I did consider that possibility @wwang0, but it adds extra decode logic and probably doesn't help much in practice.

wwang0, 2019-06-17:
That is ok. thanks

PR #216 (2019-06-28): Weaken constraint on VMADC overlap; apply new constraint to all comparisons
Closes #215
closed 2019-06-28 by kasanovic