riscvarchive/riscv-v-spec #217: the maximum element number of vrgather instruction

jovireal, 2019-06-17:
In 17.5, vector register gather instruction:
it says the source vector can be read at any index<VLMAX regardless of vl.
but then a note writes: When SEW=8, only vector elements 0-255 can be referenced.

But when SEW=8, and LMUL=8, VLMAX should equal VLEN, not limited to 256?

Is there some sort of limitation to this gather instruction?

aswaterman, 2019-06-17:
The problem is that, when SEW=8, vector elements can only represent the values 0..255.  Since the vrgather indices come from vector elements, it's not possible to access elements past 255.

Perhaps the note is confusing because it implies SEW=8 is special.  It's not.  In general, vrgather is limited to gathering elements 0..min(vl-1, 2^sew-1).

jovireal, 2019-06-17:
Oh i see, thanks~

closed 2019-06-17 by jovireal