riscvarchive/riscv-v-spec #218: The vl value of masked Unit-stride Fault-Only-First Loads instructions

wwang0, 2019-06-18:
When instruction like vlbff.v is masked(vm=0) and it takes a trap on 0, what value will be set for vl:  the length of unmasked elements, or the length of masked + unmasked elements? It should be latter from v-spec convention, but need double check here.

BTW, what is the value of masked version of vlxff intructions?

aswaterman, 2019-06-18:
With the precise-exception model, elements appear to be processed in-order.  So if a first-only-fault load traps on element 0, no later elements should be processed, and therefore it doesn't matter whether the subsequent elements are masked off.

wwang0, 2019-06-18:
I didn't make myself clear. Here I didn't mean the subsequent element after element 0. I mean the masked elements before element 0. now I have no doult vl should be the length of masked + unmasked elements (before 0), but I am curious if there is any scenario using masked version of vlxff intructions.

kasanovic, 2019-06-28:
Masking is orthogonal to vl*ff. 
`while (1) {if (a[i]>3) b[i]=c[i]; if (a[i]>7) break; i++}`
a[] can be loaded using unmasked vl*ff, c[] can be loaded using masked vl*ff.
The c[] vector could be near a protection boundary and a[] vector might not be (i.e. cannot rely on vector length trimming of first a[] load to prevent trap on c[] vector load.). Note, code would have to check break condition before storing to b[].

wwang0, 2019-06-28:
Good to know. Thanks @kasanovic.

closed 2019-06-28 by wwang0