riscvarchive/riscv-v-spec #220: imm/rs1 should be sign extended or zero extended?

jovireal, 2019-06-19:
The imm source in vector instructions might need to be extended to SEW, but whether to sign extend or zero extend is not mentioned in some of the instructions, such as vmerge.vi.

Is there a default extending mode for these instructions?

Also, in unsigned widening instructions like vwaddu.vx when XLEN<SEW, should rs1 be zero extended or, as mentioned in 11.1, sign extended to SEW? Then treat it as an SEW-bit unsigned integer?

jim-wilson, 2019-06-19:
The binutils port assumes that everything is sign-extended except instructions that use the immediate as a shift, and these instructions should already explicitly state that the immediate is unsigned.

Note for instance in the unsigned compare section, it mentions that the immediate range includes ~0 (-1) to ~0-15 (-16), because the immediate is sign-extended.

jovireal, 2019-06-20:
Thanks～I have checked the part you said and try to understand it. 
But  i'm still a little confused. It says the effective immediate range is 0 to 15 for vmsleu.vi. Does that mean when using and encoding this instruction, the MSB of the 5-bit immediate should always be zero? 

I really hope that the extending mode could be more clearly defined and explained in the spec.

aswaterman, 2019-06-20:
The text for `vmsleu.vi` actually says `0 to 15 (and (~0)-15 to ~0)`, i.e., it is sign-extended even though the instruction is unsigned.

I agree the spec should be more explicit about this, but the general pattern is that immediates are sign extended, unless they represent shift amounts or element indices (e.g. the slide instructions, whose pseudocode use the token `uimm` instead of `imm`).

jim-wilson, 2019-06-20:
It says "The vmsleu.vi range is 0 to 15 (and (~0)-15 to ~0), giving an effective vmsltu.vi range of 1 to 16 (Note, vmsltu.vi with immediate 0 is not useful as it is always false)."

Note that ~0-15 to ~0 in twos complement is -16 to -1, so the full range for vmsleu.vi is -16 to 15 as expected.  The doc is just stating it in an odd way to avoid using negative numbers with an unsigned operation.  It then says the effective range for vmsltu.vi is 1 to 16.  By "effective" it means "useful".  The full range is -15 to 16 as expected.  But 0 is not useful as it is always false, and -15 to -1 are unlikely to be useful in real code

aswaterman, 2019-06-20:
What Jim said.

jovireal, 2019-06-20:
Thank you for your answers. I can see that the basic rule is sign-extending the immediate.

Since It's true that negative immediate is not useful in unsigned instructions, I just think that it might expand the representable range of the 5-bit immediate when it is treated as unsigned integers in unsigned instructions. 
And also, for unsigned compare with scalar register rs1, when XLEN=SEW, rs1 is treated as unsigned integer to compare with vs2[i], right? But when XLEN< SEW, it might need to be sign-extended to SEW, then what is the result compared with vs2[i[?

aswaterman, 2019-06-20:
For many unsigned instructions, the signed immediate is much more useful
than additional unsigned range.

Re: rs1, see Sec. 11.1: "If XLEN<SEW, the value from the x register is
sign-extended to SEW bits."

jovireal, 2019-06-20:
Oh,  that's interesting～
I really hope there would be more detailed  illustration about this in the spec to help users understand the design and its usage.

wwang0, 2019-06-28:
@aswaterman , can you give me an example about:
"
For many unsigned instructions, the signed immediate is much more useful
than additional unsigned range.
"
Thanks

aswaterman, 2019-06-28:
This shows up with bitwise logical ops: using ANDI to mask off the LSBs of a number, or using XORI to complement a number, requires a sign-extended immediate.

JamesKenneyImperas, 2019-08-08:
To summarize the discussions above, instructions with immediate arguments treat them as follows:

**two ranges, 0...15 and ~0-15...~0**
vmsleu.vi
vmsgtu.vi
vsaddu.vi

**one range, 0...31**
vnclipu.vi
vnclip.vi
vsll.vi
vsrl.vi
vsra.vi
vnsrl.vi
vnsra.vi
vssrl.vi
vssra.vi
vslideup.vi
vslidedown.vi
vrgather.vi

**one range, -16...15**
vadd.vi
vrsub.vi
vadc.vim
vmadc.vim
vand.vi
vor.vi
vxor.vi
vmseq.vi
vmsne.vi
vmsle.vi
vmsgt.vi
vmerge.vim
vmv.v.i
vsadd.vi
vaadd.vi

The three instructions in the "two range" group are the ones to be aware of because their behavior is not intuitive.

Is this correct?

aswaterman, 2019-08-08:
That's true but for the wrong reason, as it overstates the complexity.  The first option (0...15 and ~0-15...~0) is exactly equivalent to the third option (-16...15).  So there really are only two options: zero-extended and sign-extended.

JamesKenneyImperas, 2019-08-08:
I fully understand the reason and that there are only two groups, not three. But the fundamental issue is that these three instructions are not intuitive in their behavior because they treat a sign-extended constant as an unsigned value. My intention was to clarify this for other readers and to check that these are the **only** instructions that do this.

aswaterman, 2019-08-08:
Yeah, I see your point.  Note the base ISA does the same thing (using sign-extended immediates in the context of unsigned operations).

kasanovic, 2019-12-07:
This could use some clarification in text, if someone wants to make a PR.

HanKuanChen, 2020-02-10:
@aswaterman @kasanovic 
I want to check the behavior of `vmsleu.vi v0, v0, -7`.
Please correct me if I am wrong.

# For SEW = 8

-7 (decimal in asm) -> 11001 (simm5) -> 11111001 (sign-extended) -> interpret as unsigned, which is 249 (decimal) -> compared 249 with v0

# For SEW = 16

-7 (decimal in asm) -> 11001 (simm5) -> 1111111111111001 (sign-extended) -> interpret as unsigned, which is 65529 (decimal) -> compared 65529 with v0

# Conclusion
`vmsleu.vi` can only accept [-16, 15] immediate value in assembly.
But `vmsleu.vi` can compare value between [0, 15] and [(1 << SEW) - 16, (1 << SEW) - 1], right?

David-Horner, 2020-02-10:
There is commentary below the definition of vmsleu that helps clarify this.

> The vmsleu.vi range is 0 to 15 (and (~0)-15 to ~0), giving an 
> effective vmsltu.vi range of 1 to 16 (Note, vmsltu.vi with immediate 0 
> is not useful as it
> is always false).
Since [(1 << SEW) - 16, (1 << SEW) - 1] is equivalent to (~0)-15 to ~0) 
then you are indeed correct.

closed 2020-03-28 by aswaterman