riscvarchive/riscv-v-spec #228: Why does not support m16 and m32 in vector register grouping?

tunghwa, 2019-06-29:
As I can tell, the following instructions are possible for using all 32 vector registers:
1] m16 multiplications such as out-vector = scalar * in-vector;
2] m16 MAC operations such as in-out-vector += scalar * in-vector;
3] m32 operation such as in-out-vector *= scalar;

aka-sps, 2019-07-01:
> A.3. Memcpy example
> vsetvli t0, a2, e8,m8

m8 - Inefficient use of the register file
m32 - full use of the register file  and four times less iterations

> A.5. SAXPY example
> vsetvli a4, a0, e32, m8

m8 - Inefficient use of the register file, 
m16 - full use of the register file and two times less iterations

What `vlmul` should be used in the following example instead of the implicit `m1`?
> A.4. Conditional example
> vsetvli t0, a0, e16

tunghwa, 2019-07-01:
A.6. SGEMM can also be coded using m8 and m16 to be much faster depending on hardware implementation (which is more costly on m16 than m8).

aka-sps, 2019-07-01:
@tunghwa 
> which is more costly on m16 than m8

If increasing the value of `vlmul` only leads to a concatenation of registers (more complete use of the register file), how does this affect the cost of the equipment?
In my opinion, this is the same operation that sequentially switches ports from one register in the group to the next register.  You cannot process several registers in a group in parallel if you literally follow current specification ;-)

tunghwa, 2019-07-01:
Assuming VLEN = ELEN, Using multiplications as example, hardware implementation of 16 in parallel requires
twice as many multipliers comparing 8 in parallel.

Using block matrix multiplication (divide-and-conquer) as example, doing m8,
      We can do CmxN = A[m][k] * BkxN using vector <C[m][n+0], C[m][n+1], …, C[m][n+7]>
      and vector <B[k][n+0], B[k][n+1], …, B[k][n+7]>
This only use 16 out of 32 vector registers. Thus, if m16 is supported, then all 32 vector registers are fully utilized.

From: Pavel S. Smirnov
Sent: Monday, July 1, 2019 9:44 AM
To: riscv/riscv-v-spec
Cc: Tunghwa Wang; Mention
Subject: Re: [riscv/riscv-v-spec] Why does not support m16 and m32 in vectorregister grouping? (#228)

@tunghwa
which is more costly on m16 than m8
If increasing the value of vlmul only leads to a concatenation of registers (more complete use of the register file), how does this affect the cost of the equipment?
In my opinion, this is the same operation that sequentially switches ports from one register in the group to the next register. You cannot process several registers in a group in parallel if you literally follow current specification ;-)

tunghwa, 2019-07-01:
@aka-sps 
Maybe I missed something: my point is that ISA spec normally should not dictate hardware designers how to implement.

tunghwa, 2019-07-01:
For maximum m8, maybe one multiplier is used to work on vector groups m1, m2, m4, and m8 one register a time for 1, 2, 4, and 8 times. If m16 is supported, two multipliers may be used to work on vector groups m2, m4, m8, and m16 two vector registers at time for 1, 2, 4, and 8 times (and take care of m1 case by using one of the two multipliers).  Thus, how many multipliers is implemented should not be dictated by ISA spec.

aka-sps, 2019-07-01:
@tunghwa 
Probably we mean different things.

Vector instructions are not executed completely in parallel. RVV is not a fully parallel SIMD.
And vector “registers” are not real atomically modified registers, but only the designation of vector operands. Only vector elements can be considered real “registers”.

If there are` K` processing elements (lanes), then a series of `N` sequential (but, probably, pipelined) waves, where `N=ceil(vl/K)`, is performed. Each wave processes up to `K` elements in parallel.

You can implement fully compatible RVV with single processing element (with `K=1`).
`m16`, `m32` are not `K`, but multiplier of VLEN (of `vl` and `N`) only.

The number of vector elements is not equal to the number of processing units (I hope).
Trying to do everything in parallel is overkill: the main limitations of the proposed architecture are not in the number of parallel processes, but in the bandwidth of the main memory and of the RVV register bank.

Of course, you can handle different registers in parallel (especially in the case of VLEN = ELEN), but you do not have to do this. 

On the contrary, in the case when one register of a group contains many elements, the next register should not be processed until you finish processing the previous register. Otherwise, the context switching time will be determined by the number of not yet processed elements in the current register: before updating `vstart` it will be necessary to fill all the gaps in such a broken vector representation (or roll back all the data in all registers except the first one in the group).

From this point of view, the “interleaving” of vector register elements is better than the “concatenation” of complete vector registers. Model with “interleaved” register elements in a group allows, for example, to use fewer instructions in complex number arithmetic.

Thank you for your proposal.
Regardless of the motivation, this proposal can make the architecture more efficient.

tunghwa, 2019-07-01:
Yes, agree. Using VLEN = 2 * ELAN, one can still implement with 2 multipliers (using the same example).
m8 can then be used to perform 16x16 matrix multiplication.
How to implement is issue of hardware designer. This is a side issue.
The original issue is why m16 and m32 are not supported.

From: Pavel S. Smirnov
Sent: Monday, July 1, 2019 11:53 AM
To: riscv/riscv-v-spec
Cc: Tunghwa Wang; Mention
Subject: Re: [riscv/riscv-v-spec] Why does not support m16 and m32 in vectorregister grouping? (#228)

@tunghwa
Probably we mean different things.
Vector instructions are not executed completely in parallel. RVV is not a fully parallel SIMD.
And vector “registers” are not real atomically modified registers, but only the designation of vector operands. Only vector elements can be considered real “registers”.
If there are K processing elements (lanes), then a series of N sequential (but, probably, pipelined) waves, where N=ceil(vl/K), is performed. Each wave processes up to K elements in parallel.
You can implement fully compatible RVV with single processing element (with K=1).
m16, m32 are not K, but multiplier of VLEN (of vl and N) only.
The number of vector elements is not equal to the number of processing units (I hope).
Trying to do everything in parallel is overkill: the main limitations of the proposed architecture are not in the number of parallel processes, but in the bandwidth of the main memory and of the RVV register bank.
Of course, you can handle different registers in parallel (especially in the case of VLEN = ELEN), but you do not have to do this.
On the contrary, in the case when one register of a group contains many elements, the next register should not be processed until you finish processing the previous register. Otherwise, the context switching time will be determined by the number of not yet processed elements in the current register: before updating vstart it will be necessary to fill all the gaps in such a broken vector representation (or roll back all the data in all registers except the first one in the group).
From this point of view, the “interleaving” of vector register elements is better than the “concatenation” of complete vector registers. Model with “interleaved” register elements in a group allows, for example, to use fewer instructions in complex number arithmetic.
Thank you for your proposal.
Regardless of the motivation, this proposal can make the architecture more efficient.

aswaterman, 2019-07-01:
There is also cost to larger LMUL (for optimized handling of tail zeroing in particular), and relatively little utility with 32 vector registers.  Implementations will be tuned to work well with LMULs smaller than 16, because that is a common case, and so things like memcpy generally won't go faster with LMUL >= 16.

The savings in instruction fetch energy is very minor past LMUL=8.   For a VLEN=128 machine with LMUL=8, memcpy fetches 20 bits of instruction to load 1024 bits and store 1024 bits.

tunghwa, 2019-07-01:
From the discussion, I do not see that zero fill is absolutely required. Hardware implementation may not
need to zero fill portion of the vector register for any element outside the range. It can simply stops at the
last element properly (so it is still an issue of hardware implementation). Even for multiple elements in one
vector register, leaving it dirty is possible if properly guarded. For example, if VLEN = 4 * ELAN, and using
m2 to load a vector of size 7,
      v[n+0]:    3   2   1   0
      v[n+1]:    7   6   5   4
If the operation is element-wise  (K = 1) for 0, 1, …, 6, you may either stop at 6 or get a dirty 7.

Is there any issue not considered by me for decision of zero fill?

From: Andrew Waterman
Sent: Monday, July 1, 2019 12:24 PM
To: riscv/riscv-v-spec
Cc: Tunghwa Wang; Mention
Subject: Re: [riscv/riscv-v-spec] Why does not support m16 and m32 in vectorregister grouping? (#228)

There is also cost to larger LMUL (for optimized handling of tail zeroing in particular), and relatively little utility with 32 vector registers. Implementations will be tuned to work well with LMULs smaller than 16, because that is a common case, and so things like memcpy generally won't go faster with LMUL >= 16.
The savings in instruction fetch energy is very minor past LMUL=8. For a VLEN=128 machine with LMUL=8, memcpy fetches 20 bits of instruction to load 1024 bits and store 1024 bits.

aka-sps, 2019-07-01:
@aswaterman 
"Tail zeroing" is a bad idea: we lose a lot of useful processing patterns.
Of course, there is no voting procedure, and you can freely reject all offers that conflict with your own implementation.

Let's reach a compromise at least on the current problem: the specification should reserve space for WARL field for `vlmul` up to `m32`, but only LMUL up to your `m8` will be mandatory for any RVV.

> Implementations will be tuned to work well with LMULs smaller than 16

Your implementation. But your implementation should not prevent other implementations from being more optimal.
Let the best architecture win.

aswaterman, 2019-07-01:
I'm injecting one opinion, just like you.  This kind of thing gets decided in the task group meetings on Fridays, not by snarky postings on the GitHub issue tracker.

tunghwa, 2019-07-02:
Sorry that I have no access to the task group. My starting this issue is because of trying to utilize all vector
registers. From the spec, no matter what the VLEN is, 32 vector registers must be supported. Unless there is a
similar E version (such as RV32I to RV32E) to reduce to 16 vector registers, m=8 normally is not enough
considering only instructions ‘out-vector = in-vector-1 op in-vector-2’ will utilize 8, 16, or 24 vectors. Maybe in
rare cases of very large functions, the other 8 vectors registers may get utilized. But then compiler is going to have
hard time to generate code using vector instructions – that is, only assembly programming is possible.

Anyway, the following code, assuming VLEN = ELEN = 32 bits, for sgemm_8x8  (with strideB == strideC but general
enough to  handle any 1 < M, N, K < 9) may offer a better example: (C code has been tested but not assembly code)

  # void sgemm_8x8(int M, int N, int K, float *A, float *B, float *C,
  #                int strideK, int strideN)
  #define M a0 // 0 < M < 9
  #define N a1 // 0 < N < 9
  #define K a2 // 0 < K < 9
  #define A a3 // MxK matrix
  #define B a4 // KxN matrix
  #define C a5 // MxN matrix
  #define strideK a6 // original K for A
  #define strideN a7 // original N for B and C

  #define ap A // pointer A is corrupted
  #define bp B
  #define cp C // pointer C is corrupted
  #define vl t0
  #define mt M // M is corrupted
  #define kt t1
  #define ap2 t2
  #define bp2 t3
  #define cp2 cp // n-loop is corrupted

  #define a fa0

  segmm_8x8:
  # stride length in bytes
          slli    strideK, strideK, 2
          slli    strideN, strideN, 2

  # sanity check can be disabled if this is only a private function
  # implying unsigned M, N, and K
          bnez    M, exit
          bnez    N, exit
          bnez    K, exit
 
  # { int m, n, k;
  #   float *ap = A; // ap = &A[0][0]
  #   float *bp = B; // bp = &B[0][0]
  #   float *cp = C; // cp = &C[0][0]
  #
  #   for (m = 0; m < M; m++)
  m_loop:
  #   { float *ap2 = ap; // ap2 = &A[m][0]
  #     float *bp2 = bp;  // bp2 = &B[0][0]
  #
  #     for (k = 0; k < K; k++)
          mv      ap2, ap
          mv      bp2, bp
          mv      kt, K
  #     { float *bp2 = bp; // bp2 = &B[k][0]
  #       float *cp2 = cp; // cp2 = &C[m][0]
  #
  #       for (n = 0; n < N; n++)
  #       {
          vsetvli vl, N, e32, m8
          vlw.v   v0, (cp2)        # load m-th row of C - C[m][0] to C[M][N – 1]

  k_loop:
          vlw.v   v16, (bp2)       # load k-th row of B - B[k][0] to B[k][N – 1]
          flw     a, (ap2)          # a = A[m][k]
  #         cp2[0] += ap2[0] * bp2[0];
  #         bp2++; // bp2 = &B[k][n]
  #         cp2++; // cp2 = &C[m][k]
  #       }
          vfmacc.vf v0, a, v16    # C[m][n] += A[m][k]* N[k][n] for 0 <= n < N
  #       ap2++;            // ap2 = &A[m][k]
  #       bp2 += strideN; // bp2 = &B[k][0]
  #     }
          addi    ap2, ap2, 4      # bump pointers
          add     bp2, bp2, strideN
          addi    kt, kt, -1       # decrement count
          bnez    kt, k_loop

          vsw.v   v0, (cp2)        # store this row of C
  #     ap += strideK; // ap = &A[m][0]
  #     cp += strideN; // cp = &C[m][0]
  #   }
          add     ap, ap, strideK # bump pointers
          add     cp, cp, strideN
          addi    mt, mt, -1       # decrement count
          bnez    mt, m_loop

  # }
exit:
          ret

From: Andrew Waterman
Sent: Monday, July 1, 2019 4:04 PM
To: riscv/riscv-v-spec
Cc: Tunghwa Wang; Mention
Subject: Re: [riscv/riscv-v-spec] Why does not support m16 and m32 in vectorregister grouping? (#228)

I'm injecting one opinion, just like you. This kind of thing gets decided in the task group meetings on Fridays, not by snarky postings on the GitHub issue tracker.

aswaterman, 2019-07-02:
If you register-block the SGEMM code, you can use the unused vector
registers to reduce memory traffic and energy consumption.  A
register-blocked implementation can get to much less than 1 load per FMA,
whereas simply increasing LMUL to do longer vectors still leaves you at >1
load per FMA.

tunghwa, 2019-07-02:
From you argument, I would strongly suggest E-version applies to all extensions such as F, D, Q, and V.

From: Andrew Waterman
Sent: Monday, July 1, 2019 6:00 PM
To: riscv/riscv-v-spec
Cc: Tunghwa Wang; Mention
Subject: Re: [riscv/riscv-v-spec] Why does not support m16 and m32 in vectorregister grouping? (#228)

If you register-block the SGEMM code, you can use the unused vector
registers to reduce memory traffic and energy consumption. A
register-blocked implementation can get to much less than 1 load per FMA,
whereas simply increasing LMUL to do longer vectors still leaves you at >1
load per FMA.

aswaterman, 2019-07-02:
Yes, there are separate plans to optionally reduce the scalar state by
merging the FP and integer regfiles, so e.g. RV32EF_Zfinx has 15 registers
total, or RV32IF_Zfinx has 31 registers total.
https://github.com/riscv/riscv-isa-manual/pull/281/files#diff-e27f916166f5188e4f6832bb24d3bc76R53

Defining a vector variant with a reduced register set is another possible
direction.

aka-sps, 2019-07-02:
### Considering current issue:
Who will prepare pull request with following changes in Table 2?

~~~~
|2:0 | vlmul[2:0]  |Vector register group multiplier (LMUL) setting
~~~~

Do you have any other suggestions?

### About "Tail zeroing"
The Friday meeting minutes only contain:
> Major changes from v0.6:
>
> zeroing for >= vl

"Tail zeroing" issue was raised repeatedly, there were different opinions (#157), but this is not reflected in the meeting minutes.

List of voting questions and results of previous voting can be very useful to avoid duplication of questions and clarify the status of previously asked questions.

IMHO: Meeting is better for clarification and discussion. The tracker is better suited for issues than meeting (and `+1`/`-1` reaction icons can be used for voting).

@aswaterman Sorry if instead of disappointment with the current state of RVV, you only saw a snide remark. I know you and your work and sincerely respect you.  Thank you for your patience and understanding.

tunghwa, 2019-07-02:
Now we are off my original topic too much. But let’s continue on it:
(1) Hint to cache pre-loading can do the job: no need to pre-load into registers.
(2) Personally, Zfinx is not a very good idea comparing to floating point register mapping in a way different from NaN boxing:
         Say we have 32 F registers,
            mapping f[2n+0] and f[2n+1] to d[n] gives 16 D registers
            optionally mapping f[4n+0], f[4n+1], f[4n+2] and f[4n+3] to q[n] gives 8 Q registers
        Then mapping them again to V registers, we can do
            up, to m16: 32 VLEN=32 V registers
            up to m8: 16  VLEN=64 V registers
            optionally, up to m4: 8 VLEN=128 V registers
       We always have a good mix of all kind of registers and save lots of hardware resource for offering K=2 instead of K=1.

Anyway, it is hardware designers’ job to come up the best system overall throughput. A spec should only offer the possibility
and left the hardware designers to find the best solution they can come up.

From: Andrew Waterman
Sent: Monday, July 1, 2019 6:14 PM
To: riscv/riscv-v-spec
Cc: Tunghwa Wang; Mention
Subject: Re: [riscv/riscv-v-spec] Why does not support m16 and m32 in vectorregister grouping? (#228)

Yes, there are separate plans to optionally reduce the scalar state by
merging the FP and integer regfiles, so e.g. RV32EF_Zfinx has 15 registers
total, or RV32IF_Zfinx has 31 registers total.
https://github.com/riscv/riscv-isa-manual/pull/281/files#diff-e27f916166f5188e4f6832bb24d3bc76R53

Defining a vector variant with a reduced register set is another possible
direction.

aswaterman, 2019-07-02:
@aka-sps Understood.  Thank you.

I was not at last Friday's meeting, but my understanding is that they did not reach a decision on eliminating tail zeroing, and someone has been tasked with writing up the pros and cons so it can be discussed over email.  (I also support removing that feature, just not very strongly so.)

As to LMUL > 8: we don't actually have to take any action now, because vlmul[2] doesn't need to be contiguous with vlmul[1:0].  It's ugly to split the field up, but it doesn't affect runtime performance as long as it can still be set with a single vsetvli instruction.  The task group should discuss this one.

kasanovic, 2019-11-10:
LMUL>8 with only 32 registers has limited utility, and any possible marginal gains in efficiency will be small for good implementations (i.e., those that optimize for all the cases where LMUL<=8 must be used to avoid register spills).

We are anticipating supporting >32 vector registers in the longer instruction encodings, in which case, LMUL>8 would make sense.  Setting aside space for a wider contiguous LMUL field is possible, but there are other possible uses of the 11-bit  immediate field that might be more important in the shorter 32-bit instruction encoding.  Discontiguous LMUL field is not a practical problem, mainly just an aesthetics issue.

kasanovic, 2019-12-07:
Tail zeroing was dropped.
We'll stick with LMUL=8 as max in 32-bit encoding with only 32 registers.
Closing this issue as it meandered across several topics.  Should stick to one topic per issue.

closed 2019-12-07 by kasanovic

aswaterman, 2019-12-07:
Note also the current masking scheme requires LMUL < SEW, so it can't support LMUL > 8 without a constraint on SEW or a redesign.