riscvarchive/riscv-v-spec #231: Move the `vm` bit from instruction to RVV state

aka-sps, 2019-07-01:
RVV instruction `vm` bit (bit number 25) probably will have low entropy.
For consecutive vector instructions, changing the value of this bit is unlikely.

To make better use of the code space, it is worth moving this bit to the general state of the vector extension.

Vector instructions take several cycles and setting the masking bit using a separate instruction should not significantly affect performance.

In addition to setting the masking mode, you can select direct or inverse (w/o `vmnot.m v0, v0`) masking mode and even the register number of the mask.

aswaterman, 2019-07-01:
It's exceedingly unusual to have a strip-mine loop with only masked instructions, as some instruction needs to generate the mask.  So this will add a vsetvl to nearly every strip-mine loop with masked instructions, and multiple vsetvls for others.  For some implementations, this won't matter for performance.  For others (spatial machines without wide superscalar issue), it will matter.  For all of them, it hurts code density.

Perhaps the greater concern is how this will interact with an expanded vector ISA with a 48-bit or 64-bit encoding.  In that case, the mask register will be a full register specifier, rather than using v0 implicitly, rendering vestigial the `vm` state bit.

aka-sps, 2019-07-01:
Let's first separate the issue of location of the masking bit and the type of masking from the issue of optional selection of the mask register. I apologize for merging the two questions into issue.

From 6 examples of Appendix A:
A.2. We need two additional `set mask` instructions per loop.
A.4. We need 3 additional instructions `set mask` (one of them replaces the extra ` vmnot.m v0, v0`)

For 16.7. (strcpy and strncpy) examples: two additional instructions for each.
For 16.8. compact_non_zero: two additional instructions.

From 9 examples we can see only 11 places where the mask changes.

Please add more examples where the masking mode changes more often.

aswaterman, 2019-11-29:
Closing this issue as there isn't consensus to make this change.  Moving the mask bit into the vector state would require two extra instructions per loop iteration when vectorizing code that contains both an if-statement and some code outside of the if-statement.

closed 2019-11-29 by aswaterman