riscvarchive/riscv-v-spec #233: Is the constant value of SLEN architecturally visible?

aka-sps, 2019-07-02:
How can I get the value of the SLEN constant?

aswaterman, 2019-07-02:
It's computable by putting a known pattern into a register group and examining the group with a different SEW/LMUL ratio.  I believe the following will put SLEN into t0.  (Note that vmv.x.s ignores LMUL, so is effectively accessing the register group with a different SEW/LMUL ratio than the vid.v instruction wrote it with.)

    vsetvli x0, x0, e32,m2
    vid.v v0
    vmv.x.s t0, v1
    slli t0, t0, 5

aka-sps, 2019-07-03:
~~~~
    /* vl=VLMAX for AVL=x0 */
    vsetvli x0, x0, e32,m2

    /* fill by 0..VLMAX-1, v1[0] is SLEN / 32  for e32 */
    vid.v v0
    
    /* vext.x.v t0, v1, x0  ignores LMUL and vector register groups, copy v1[0]=SLEN / 32 to t0 */
    vmv.x.s t0, v1

    /* t0 is SLEN */
    slli t0, t0, 5
~~~~

It seems OK.

Very inconvenient model for programmers.
For a programmer, it would be more convenient to fix SLEN to min SEW, or to VLEN.

Why SLEN ≥ 32 but not SLEN ≥ 8 (min SEW) or, for example SLEN ≥ ELEN?

aswaterman, 2019-07-03:
It shouldn't be inconvenient for programmers in practice.  Following the code-generation rule that the SEW / LMUL ratio be held constant for a strip-mine loop, SLEN should be invisible to the programmer.

@kasanovic would have to comment on why SLEN ≥ 32.

aka-sps, 2019-07-03:
It would be nice to give an example for

VLEN=256b, SLEN=64b, SEW=128b, LMUL=4

to illustrate

> If SEW > SLEN, the striping pattern places one element in each vector register in the group before moving to the next vector register in the group. So, when LMUL=2, the even-numbered vector register contains the even-numbered elements of the vector and the odd-numbered vector register contains the odd-numbered elements of the vector.

in addition to the following examples.

>4.2. Mapping with LMUL > 1
> VLEN=256b, SLEN=128b, SEW=256b, LMUL=4
> VLEN=256b, SLEN=128b, SEW=32b, LMUL=8

aka-sps, 2019-07-03:
Do not disturb @kasanovic, please. He is probably very busy.
But such questions may arise not only from me.

I understand the case of SEW > SLEN, but an example would be useful to others.
Thank you.

kasanovic, 2019-12-07:
@ubc-guy offered to improve SLEN text in issue #329.  Closing here.

closed 2019-12-07 by kasanovic