riscvarchive/riscv-v-spec #234: SEW-bit * SEW-bit multiply result

aka-sps, 2019-07-02:
> 12.13. Vector Widening Integer Multiply-Add Instructions
> The widening integer multiply-add instructions add a SEW-bit * SEW-bit multiply result to (from) a 2 * SEW-bit value and produce a 2 * SEW-bit result.

For example, if SEW=32 then SEW * SEW is 1024-bit that greater than 2 * SEW=54.
What you mean exactly?

1. Multiply a SEW-bit to a SEW-bit with a SEW-bit multiply result and add to a 2 * SEW-bit accumulator
2. Multiply a SEW-bit to a SEW-bit with a 2 * SEW-bit multiply result and add to a 2 * SEW-bit accumulator

kasanovic, 2019-07-02:
Option 2, as the text say (2*SEW-bit value).

closed 2019-07-02 by kasanovic

aka-sps, 2019-07-02:
The fact of the matter is that the text says:
> SEW-bit * SEW-bit multiply result

Only the size of the last operand of the sum and the size of the final result are determined, but the size of the intermediate multiply result is only implied, but not explicitly defined.

Please determine _exactly and explicitly_ that the size of the intermediate multiply result is 
SEW + SEW = 2 * SEW bits,
not SEW bits (as in C multiply operation and in the `mul` instruction), 
and of course not SEW * SEW bits, as can also be interpreted from the text.

Thanks