riscvarchive/riscv-v-spec #235: reciprocal and reciprocal square-root approximations

aswaterman, 2019-07-03:
Some other architectures offer instructions that approximate 1/n and 1/√n.  These aren't replacements for IEEE 754 div/sqrt, but are meant to accelerate these operations when some imprecision is OK.  For example, in addition to 754-compliant operations, Intel SSE provides instructions that provide approximations to 12 bits of precision.  (With 12 bits, you can get almost single precision with one Newton-Raphson iteration, and most of the way to double-precision with a second iteration.)

We should consider whether or not to include operations like this.  (I'm not taking a strong position; I'm just making sure we consciously accept or reject them.)

brucehoult, 2019-07-25:
I've always been surprised that RISC-V didn't offer these instructions.

I'd also like to see instructions to extract the exponent (with extended range for an unnormalised argument), and to add an integer to the exponent of an FP number, correctly handling denormalisation and overflow to infinity. These are *extremely* helpful for implementing range reduction and so forth in transcendental library functions.

12 bits is a lot and I think the difficulty/table size is not justified. The main benefit of these instructions is handling special cases and adjustment of the exponent.

PowerPC 603 and 604 and later had scalar instructions with 5 bits of accuracy. I couldn't immediately locate what Altivec has.

ARMv8 returns a value with significand in the range 256..511. Pseudocode can be found at http://shell-storm.org/armv8-a/ISA_v85A_A64_xml_00bet8/xhtml/shared_pseudocode.html#impl-shared.FPRSqrtEstimate.2

ARMv8 also has the Newton-Raphson steps for recip and rsqrt as vector instructions. That may be going too far.

aswaterman, 2020-05-31:
I've surveyed the space some more.  SSE seems lonely in choosing 12 bits.  Roughly 8 bits seems to be the most popular choice.  In addition to ARM, Itanium also chose roughly 8 bits of precision.  Recent versions of POWER use 8 bits of precision (for reciprocal only).

The popularity of 8-or-so bits is not too much of a surprise.  For reciprocal square root, relative error of 2^-7 is just barely good enough to for two iterations to get within two ulps of single-precision.  But it's not quite good enough for three iterations to get within two ulps of double-precision; the relative error needs to be closer to 2^-7.4 for that.  (Evidence forthcoming.)  The story is similar for reciprocal.

As far as cost goes, to reach a relative error bounds of of 2^-7.4 or so, the lookup tables take around 800 gates (in total for recip and rsqrt, using yosys' default setup).  To exceed 2^-12, it's more like 20K gates.  These numbers should obviously be taken with a grain of salt, but assuming they're reasonable to first order, then the 12-bit variant is hard to stomach.

Aside from the choice of precision, there are a couple other definitional issues to address:
- Exceptions, subnormals, etc.: the consensus among other ISAs is to handle these correctly.
- Implementation-defined behavior: it's obviously possible to force all implementations to implement these instructions in the same way, but I'm inclined to define them in terms of maximum relative error, to provide more implementation flexibility.  For example, when vfrecip performance isn't important, it could be implemented by dividing into 1.0, using the vfdiv datapath.

ebahapo, 2020-06-02:
Hi, Andrew.

> I've surveyed the space some more. SSE seems lonely in choosing 12 bits. Roughly 8 bits seems to be the most popular choice. In addition to ARM, Itanium also chose roughly 8 bits of precision. [...]
Having worked before on using these instructions in compilers on x86-64 and AArch64, I appreciated the extra precision provided by 12 bits in SSE then.  The series would converge quickly and achieve a good enough precision for some popular benchmarks (e.g., SPECCPU), albeit only for single precision at the time.  My only experience with 8 bits in NEON (or was it 9 bits?) was less satisfactory, as the series required an additional iteration for single precision and two extra iterations for double precision, if memory serves.  Also, the double precision reciprocal (1/x) would not converge to acceptable precision for commercial benchmarks, even with extra iterations, to be useful in general.

In terms of performance, of course it depended on how the series compared with the respective division and square root instructions, not only in terms of latency, but also of throughput.  Algorithms may be throughput bound or latency bound on these operations, and, unfortunately, neither LLVM nor GCC are capable of identifying them.

Therefore, the initial number of bits may depend on the intended application.  Graphics is a notorious application of such instructions, when the precision is less important.  HPC however requires throughput and precision.  So I'd favor the latter choice of definition, provided that this information is available to user code for use by libraries.

HTH

__ 
Evandro Menezes ◊ SiFive ◊ Austin, TX

aswaterman, 2020-06-02:
Thanks for the input @ebahapo.  Of course, the tradeoff mostly boils down to whether the extra iteration accounts for enough overall runtime to justify the cost increase (bearing in mind that these instructions will probably get no play for code not compiled with -ffast-math).

> two extra iterations for double precision

For double precision, I think it's either +1 or +0, depending on how much error you can tolerate.  The convergence (very approximately) goes 12->24->48->nearly-double-precision if you start with 12 bits, or 8->16->32->nearly-double-precision if you start with 8 bits.  So if several ulps is OK, then the 12-bit approximation is one iteration faster.  If only a few ulps is OK, then the two schemes perform the same.

So I think the extra iteration for single-precision and half-precision are the bigger concerns.

kasanovic, 2020-06-26:
Andrew contributed the following design proposal on mailing list:
The task group has recommended moving forward with adding instructions that estimate reciprocals and reciprocal square roots.  These are both useful for -ffast-math code where it's acceptable to sacrifice a bit of precision in exchange for greater performance.  Reciprocal square root in particular is ubiquitous in graphics and physics computations.  These instructions can feed into Newton-Raphson schemes to provide refined estimates of quotients and square roots.

Following is a discussion of some design considerations leading to a concrete proposal.

Strict or bounded definition?

Some ISAs have required these instructions compute the same estimate on any implementation, whereas others have only specified a maximum error bound.  A strict definition is obviously preferable for software portability and compliance testing.  A bounded definition allows more implementation flexibility: for example, implementations with a fast divider could compute the reciprocal estimate as (1.0 / x).

The task group has recommended adopting a strict definition.  I'll supply C source code that models the instruction semantics and prints Verilog code that implements the lookup-table components of the instructions.

What precision to choose?

Adopting a strict definition increases the pressure to keep the cost low, since some applications won't benefit from these instructions, but implementations are still effectively required to provide them.  So, I'm recommending using a 7-bit scheme, which keeps the cost down (hundreds of gates to implement as a LUT) while providing just enough precision for Newton-Raphson refinement to converge reasonably quickly.

A common goal is to produce an estimate within a couple ulps of the IEEE 754 result.  With that in mind, a few precision sweet spots emerge:

- 4-bit estimates (which, due to careful rounding, actually have relative error of 2^-4.6 for reciprocal and 2^-4.5 for rsqrt) take 1, 2, 3, and 4 iterations to get within a couple ulps of bfloat16, binary16, binary32, and binary64, respectively.
- 5-bit (2^-5.5, 2^-5.3) take 1, 1, 3, and 4 iterations, respectively.
- 6-bit (2^-6.4, 2^-6.3) take 0, 1, 2, and 4 iterations, respectively.
- 7-bit (2^-7.4, 2^-7.3) take 0, 1, 2, and 3 iterations, respectively.
- 9-bit (2^-9.4, 2^-9.2) take 0, 0, 2, and 3 iterations, respectively.
- 12-bit (2^-12.4, 2^-12.3) take 0, 0, 1, and 3 iterations, respectively.

All of these are local optima; the global optimum obviously depends on your favorite FP format and the ubiquity of these calculations.  The 7-bit scheme is of interest because it's the cliff at which double-precision needs 3 iterations to converge.

Instruction encoding?

As unary instructions, these instructions fit conveniently in the VFUNARY1 opcode, next to the existing VFSQRT instruction.

If, over time, more precise estimates are required, they can be accommodated in the same unary-operation encoding space.

Corner-case details?

For values outside the domain, or near its edge, these instructions should behave like you'd expect 1.0/x or 1.0/sqrt(x) to behave on an IEEE 754-compliant C implementation.  In particular:

- rsqrt(+inf) = +0
- rsqrt(+/- 0) = +/- inf, raising DZ exception
- rsqrt(nonzero negative) = qNaN, raising NV exception
- rsqrt(NaN) = qNaN, raising NV for sNaN inputs
- rsqrt(subnormal) is handled correctly; the result is normal and never overflows

- recip(+/- inf) = +/- 0
- recip(+/- 0) = +/- inf, raising DZ exception
- recip(NaN) = qNaN, raising NV if signaling
- recip(subnormal) is handled correctly, noting that all but the largest subnormals will overflow to infinity, raising OF the exception

Newton-Raphson iteration instructions?

Other ISAs have provided instructions that perform part of a Newton-Raphson iteration step, e.g., computing (3 - a * b) / 2.0 in a single instruction.  These instructions would avoid needing to splat a constant to a vector register, but I think otherwise they will only save one instruction per Newton-Raphson loop, not one instruction per Newton-Raphson iteration.  (This is because the expression can be rewritten as (3/2 - a/2 * b), where a/2 is the same for all iteration steps.)

To keep the ISA simple, and to avoid eating up opcode space, I'm recommending against adding these iteration instructions.

kasanovic, 2020-07-19:
Tentative encoding space allocation for reciprocal and reciprocal-square-root step instructions would be in the vector-vector encoding next to vfrdiv and vfrsub respectively (OPFVV, funct6=100001 and 100111).  The TG has not decided on form of these, or if they should be included in v1.0.

kasanovic, 2020-09-04:
The specification for these has now been added in commit 696966b3d74bb8f9799bf82f69134865d4d3f3b4
 Please raise any further issues on these separately.

closed 2020-09-04 by kasanovic