riscvarchive/riscv-v-spec #236: Should we have float version of Vector Slide1down and Slide1up

wwang0, 2019-07-03:
Now we only have integer slide1down and slide1up? shouldn't we support float version of slide1down and slide1up?

aswaterman, 2019-07-03:
Just for historical context, the original motivation for vslide1down was to enable an external debugger to populate a vector register without additional memory storage.  Providing a fast path to slide a vector and insert a scalar was a useful side effect.

With that said, if the vslide1 instructions are sufficiently useful in practice, I support adding FP versions.  There should be no additional datapath cost, given the existence of the vfmv.s.f instruction and the existing integer vslide1 instructions.

wwang0, 2019-07-03:
Recently I start to write Livermore Loops assemble code, and for the kernel 2: ICCG excerpt (Incomplete Cholesky Conjugate Gradient), it has loop as below:
    for ( l=1 ; l<=loop ; l++ ) {
        ii = n;
        ipntp = 0;
        do {
            ipnt = ipntp;
            ipntp += ii;
            ii /= 2;
            i = ipntp - 1;
#pragma nohazard
            for ( k=ipnt+1 ; k<ipntp ; k=k+2 ) {
                i++;
                x[i] = x[k] - v[k  ]*x[k-1] - v[k+1]*x[k+1];
            }
        } while ( ii>0 );
    }
And for the x and v which are float, we definitely can use vslide1 instructions here.

closed 2019-08-15 by wwang0

vfdff, 2022-01-25:
simpilfied case https://godbolt.org/z/r7Mb66ez5

aswaterman, 2022-01-25:
Float versions were added to the final V spec; see https://github.com/riscv/riscv-v-spec/pull/402