riscvarchive/riscv-v-spec #240: vslide1up and vslide1down behavior

go-sc, 2019-07-04:
The text description of `vslide1up` does not define behavior in case when `vstart` is non-zero.
The pseudocode is confusing in case `0 = i = vstart`:

```
 vslide1up behavior

                    i < vstart  unchanged
                0 = i = vstart  vd[i] = x[rs1] if mask enabled, unchanged if not
  max(vstart, 1) <= i < vl      vd[i] = vs2[i-1] if mask enabled, unchanged if not
              vl <= i < VLMAX   tail elements, vd[i] = 0
```

Same for `vstart <= i = vl-1` case in `vslide1down`:

```
   vslide1down behavior

                       i < vstart  unchanged
             vstart <= i < vl-1    vd[i] = vs2[i+1] if mask enabled, unchanged if not
             vstart <= i = vl-1    vd[vl-1] = x[rs1] if mask enabled, unchanged if not
                 vl <= i < VLMAX   tail elements, vd[i] = 0
```

kasanovic, 2019-11-10:
Which case is not given or ambiguous?  This seems complete and unambiguous to me.

aswaterman, 2019-11-29:
Closing as stale.  Please reopen if there is actually an ambiguous case.

closed 2019-11-29 by aswaterman