riscvarchive/riscv-v-spec #241: Need a couple of confirmation about vlxff intructions

wwang0, 2019-07-05:
suppose the data in memory is 'abc\0', and after vlbff is used to load:
1. vl=3, or vl =4?
2.  the tail element( VLMAX>index >= vl) also need to be zero'd, right?
3. When an element > 0 raises an exception, the spec says :" If an element > 0 raises an exception, that element and all following elements in the destination vector register are not modified, and the vector length vl is reduced to the number of elements processed without a trap. " it is not aligned with common standard of the spec:
    a. tail elements are not zero'd.
    b. vstart is not used, we choose to modify vl.

aswaterman, 2019-07-05:
The data values do not affect the behavior of the instruction.  '\0' has no special significance.

wwang0, 2019-07-05:
??? I don't get it, vlbff will trap when it meets 0, and then vl will be updated accordingly. my understanding is wrong?

aswaterman, 2019-07-05:
Yeah, I think there is a misunderstanding. “These instructions execute as a
regular load except that they will only take a trap on element 0.” The data
value doesn’t matter.

The strlen code example shows how to use these instructions when the goal
is to detect a 0 data value:
https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#77-unit-stride-fault-only-first-loads

wwang0, 2019-07-05:
What does 'element 0' mean? doesn't it mean the data value is 0?

aswaterman, 2019-07-05:
It means "the zeroth element in a vector", not "an element whose value is 0".

The way the instruction works is:

- If no exceptions are encountered, it behaves the same as a normal unit-stride load.
- If an exception is encountered on element number i, and i > 0, then vl is set to i, and no exception is taken.
- If an exception is encountered on element number 0, then the exception is taken.

wwang0, 2019-07-07:
The instruction really puzzled me for a long time. Can I know why we have such instruction?　I also don't quite understand the example fo strlen,  the example has chance to access memory elements after '\0', which should be illegal and might trigger memory access trap(or may not),  how does trap hanlder of system decide the trap is no harm or harm?

aswaterman, 2019-07-07:
The instruction supports vectorizing loops whose trip count isn't known at entry (i.e. `while` loops), with `strlen` being a good example.

In the `strlen` example, the `vlbff` instruction might access elements past `\0`, but it will never trap on elements past `\0`, so the trap handler does not need to do anything special.  This works because the instruction's semantics are to reduce `vl` rather than taking the exception.

wwang0, 2019-07-07:
I see. then my last question is why the exception of element 0 is taken?　Thanks!

aswaterman, 2019-07-07:
Faulting on element 0 handles the case that a page fault _should_ occur, e.g., because the middle of the string has been paged out.  Consider these two examples (assuming VLMAX = 32):

- String `abc\0` at address 0x1ff0; the page at 0x2000 is paged out.  The first `vlbff` will attempt to load bytes 0x1ff0 to 0x200f.  Since loading 0x2000 would cause an exception, `vl` is reduced to 16 (== 0x2000 - 0x1ff0) and no exception is taken.  Because a 0 byte was detected, we exit the strip-mine loop, and no exception occurs.

- Same scenario, except the string is `abcdefghijklmnopqrstuvwxyz\0`, so part of the string is paged out.  On the first iteration, `vl` is reduced to 16, but since no zero byte was found, the loop will continue.  On the second iteration, the exception is detected on the zeroth element, so the exception will really be taken this time.  Once the page fault is handled, the loop will find the zero byte on that same loop iteration, and we'll exit the loop.

Hopefully those two examples help.

wwang0, 2019-07-07:
Now I understand it completely. Thanks for your patience to explain it to me!

closed 2019-07-07 by wwang0

aswaterman, 2019-07-07:
Cool!

wwang0, 2019-07-07:
I probably have the real last question, in v spec, it says:
"
If an element > 0 raises an exception, that element and all following elements in the destination
vector register are not modified. 
"
But we usually zero the tail element, but not keep them not modified, right?

aswaterman, 2019-07-07:
Let’s postpone that question for now, as tail-zeroing semantics are still being debated.