riscvarchive/riscv-v-spec #243: Any easy way to  find location of first minimum in array?

wwang0, 2019-07-09:
the c code is like this:
     m = 0;
     for ( k=1 ; k<n ; k++ ) {
         if ( x[k] < x[m] ) m = k;
     }

what I can think of to achieve it by using vector instructions are like below:
vfredmin.vs v1, v0, v0
vfmv.f.s fa0, v1
vmfeq.vf v2, v1, fa0
vfirst.m a1, v2

but I think it is too complicated. Should we create an instruction to achive it?

aswaterman, 2019-07-09:
The slowest part is likely to be the vfmv followed by the vmfeq, since the
vfmv will cause a loss of decoupling on many implementations. That can be
avoided by splatting the reduction result using vrgather.vi v2, v1, 0,
prior to comparing. Same number of instructions, but likely higher
performance.

If we wanted to speed up this case further, we could add comparison
instructions that take scalar arguments from vector registers, rather than
scalars. That would get rid of the splat. But it would probably require a
larger instruction encoding space.

wwang0, 2019-07-10:
Got your point BTW, is there any chance we create a new instruction like vfminindex, and get index of min directly?

aswaterman, 2019-07-10:
It adds some cost to the reduction hardware to propagate indexes as well as data, so there needs to be a strong quantitative case.  My intuition is that, in the broader application context, the min-reduction + splat + compare approach won't contribute too much to runtime.

wwang0, 2019-07-10:
Thanks Andrew. Out of curiosity, do we have a document to introduce quantitiative cases for uncommon riscv-v instruction? I don't think I can get all such information in our v-spec.

aswaterman, 2019-07-10:
There's no formal process in place.

wwang0, 2019-07-10:
Thanks Andrew. Can you help me fully understand "the vfmv will cause a loss of decoupling on many implementations"?　that will also help me understand why we use element 0 of a vector register but not a scalar register to do reduction.

aswaterman, 2019-07-10:
In some implementations, the scalar processor is decoupled from the vector
processor.  The scalar processor runs ahead, and it can prefetch vector
loads for the vector unit.  The vector instructions are placed in a queue,
and the vector unit processes them when there are no hazards.  This scheme
works well, unless the scalar unit depends on data from the vector unit.
In that case, the scalar processor can no longer run ahead; it must wait
until the data from the vector becomes available.  This event is called a
'loss of decoupling'.

wwang0, 2019-07-10:
Totally understand now. Thanks!

kasanovic, 2019-12-07:
This case is complicated by need to handle FP special values correctly (e.g., is all NaN).

kasanovic, 2020-03-28:
This can be handled in software at reasonable speed with existing instructions.

closed 2020-03-28 by kasanovic