riscvarchive/riscv-v-spec #245: is vstart sufficient for segment loads and stores?

solomatnikov, 2019-07-10:
Because segments are not aligned they can cross page boundary, so segment can cause 2 page faults.

aswaterman, 2019-07-10:
Maybe you saw my PR (https://github.com/riscv/riscv-v-spec/pull/244), but I have been thinking about this issue, too.  vstart is sufficient only if we are willing to allow imprecision at the sub-structure level.  So you might load/store half a structure, then fault.  After handling the fault, you'll redundantly perform the first access again.  (Then maybe you'll fault again on the second half, and after handling that one, you'll redundantly perform both accesses again...)

I think this is OK, provided the overlap constraint mentioned in that PR.  It's better than supporting intra-segment restart, which would add additional complexity to these already-difficult instructions.

It's also consistent with how we had planned to define scalar load-multiple/store-multiple instructions, if we ever did so: these instructions would always restart at the beginning.

All of this implies that segmented accesses and load-multiple/store-multiple should only be used on idempotent memory regions.

kasanovic, 2019-12-07:
I think this was settled, so closing the issue.

closed 2019-12-07 by kasanovic