riscvarchive/riscv-v-spec #252: What is the value of VM bit encoding for mask-register logical instructions?

lihanglin, 2019-07-17:
Not only the instructions in chapter 16.1 but also 17.6 (vcompress.vm) did not support masking.
The spec. did not specify what the VM bit exactly is for those instructions.
Was it treated as Don't Care or Should be One/Zero?

jim-wilson, 2019-07-17:
Unmasked instructions are always VM=1, but the docs could perhaps explicitly say that like it does for a few others.  When an instruction doesn't support masking, VM=0 should be a reserved encoding.  The docs do say this for a few instructions.  This could probably be improved to be more consistent across all instructions that don't support masking.

lihanglin, 2019-07-17:
@jim-wilson 
Agree with you

closed 2019-11-15 by kasanovic