riscvarchive/riscv-v-spec #254: Zero-/sign-extension of vmv.x.s and vmv.s.x

aswaterman, 2019-07-19:
These instructions are currently defined to zero-extend when the destination is wider than the source.  We should make sure this is the right decision.  One important case to consider is that the ABI keeps 32-bit integers sign-extended when passed in RV64 x-registers, so moves of SEW=32 to an x-register on RV64 will often need to be sign-extended.

We can, of course, provide both variants.  They're unary instructions, so they take up little encoding space.

aswaterman, 2019-07-31:
This brings to mind a tangential issue: in the half-precision scalar extension, what should be the extension behavior of the hypothetical FMV.X.H instruction?  FMV.X.W always sign-extends its result in RV64; the primary motivation was to maintain the ABI property that all 32-bit types are sign-extended when held in 64-bit x-registers..  Sign-extending FMV.X.H seems like the right thing to do for consistency, though there isn't a strong ABI justification one way or the other.

Following that rationale (RV64 ABI in the 32-bit case, and consistency in the other cases), I am going to propose we change these instructions to sign-extend in all cases.  I will submit a PR.

kasanovic, 2019-09-03:
Closed with  pull request #272, and scalar issue moved to scalar ISA spec.

closed 2019-09-03 by kasanovic