riscvarchive/riscv-v-spec #256: signed or unsigned shift for vaadd and vasub

jovireal, 2019-07-19:
For vaadd and vasub, should the sum be treated as signed data or unsigned data before right shift? It may produce different result.

jovireal, 2019-07-23:
For example, if the result is 0xff, it means -1 in signed data and maximum value in unsigned data，and the >>1 bit operation should be clarified. Or the default setting is signed data?

aswaterman, 2019-07-23:
It's actually not about the signedness of the shift; it's about the signedness of the addition.  But your point stands.  For 8-bit numbers, the unsigned average of 0 and 0x80 is 0x40, whereas the signed average of 0 and 0x80 is 0xc0.

I assume given the lack of a `u` suffix that these are signed instructions, but I wonder if the omission of the unsigned variants is deliberate.

kasanovic, 2019-10-04:
These were intended to be signed.  Adding unsigned variants might be justified.

kasanovic, 2019-11-15:
Task group agreed to add both signed and unsigned variants.

commit c2f3157e (2019-11-15): Added signed and unsigned versions of averaging add and subtract.
Moved opcodes over to middle column in encoding table and dropped immediate forms.
Closes #256
closed 2019-11-15 by kasanovic