riscvarchive/riscv-v-spec #257: Are Fault-Only-First Loads necessary?

fintelia, 2019-07-19:
The stated reason for including these instructions is to vectorize while loops with "data dependent exit conditions". My understanding is that the concern is that using the normal variants would could the loop to potentially trigger read past the end of the array and trigger a page fault. However since page tables have a granularity of at most 4KB, couldn't the compiler just have each loop iteration clamp vector length so that accesses touch only a single page in memory? I don't imagine we'll have >32768-bit vector registers any time soon. 

The security analysis provided in the spec is also somewhat underwhelming right now. For instance, I'm not convinced that it is harmless to let user code to probe for guard pages... at very least it seems like an attractive gadget for return-oriented programming. It also doesn't address the possibility of S-mode code using it to probe for PMP regions or for VS-mode code to probe the mappings of the second level page table (in both cases having write access to page tables negates the contiguous access limitation).

aswaterman, 2019-07-19:
I agree the security concerns must be more carefully thought through.  Here are some reasons in favor of not getting rid of these instructions:

- Not all ABIs can expose a generous protection granule of 4K.  Systems without paging (for which vectors will be very popular!) can have much finer-grained PMAs or PMPs.
- I tried coding up `strlen` this way and it added 4 instructions.  That's not a problematic code size increase for a library routine, but it will disincentivize autovectorization.  It also increases the tradeoff point at which vectorization becomes profitable.
- `strlen` is the easy case.  When there's more than one data structure being operated on, the problem gets worse (maybe intractable in some cases).

fintelia, 2019-07-21:
For `strlen` I think you can get it down to two extra instructions per loop iteration (`vl = 0 - (addr | -4096)` instead of the `vl = x0` used now). Of course you could avoid any extra cycles at all by adding a new instruction. But as you point out none of that matters if you have to support protection granularities much less than one page. 

One possible compromise would be for these instructions to actually trap on an ISA level but define the ABIs to have them act as currently defined. That way an OS could transparently swap pages to disk, update accessed bits, etc. without user code knowing, but compilers would still be able to do all the vectorization optimizations as before. It would also give an escape hatch if it turned out privileged code needed to detect/prevent egregious uses of these instructions to probe the address space.

kasanovic, 2019-10-04:
Consensus is that privileged architecture can always further constrain behavior without breaking unprivileged ABI.

closed 2019-10-04 by kasanovic

dzaima, 2023-07-04:
Of note is that, as the spec permits the new `vl` to be arbitrarily lower than what's required to fulfill not faulting, an implementation could choose to do the bounding to within a 4K page by itself (the page containing the last byte of the first element), thus not ever leaking any accessibility information outside of what's observed by the first element.

Thus you'd get the security of page-aligning, and the vectorization cost of a single instruction. (with the side-benefit of aligning the pointer after crossing any page, potentially increasing perf where maximally-aligned loads are faster!)