riscvarchive/riscv-v-spec #259: vector reduction when the element 0 is masked

wwang0, 2019-07-24:
Suppose teh first element is masked in a reduction instruction, I need to confirm:
1. the reduction result still stores in element 0 of destination register;
2. vs1[0] and vs2[0] don't participate reduction. 

And another corner case, suppose all elements are masked. the element 0 of destination register is 0, or unchanged?

aswaterman, 2019-07-24:
For these instructions, the mask only controls which elements participate in the reduction.  vd[0] is written with the result even if all elements are masked.  (In that case, vs1[0] is simply copied to vd[0].)

vs1[0] *does* participate in the reduction, irrespective of the mask.

vs2[0] participates in the reduction if element 0 is not masked.

wwang0, 2019-07-24:
Got it, thanks. And the spec needs to clarify it.

aswaterman, 2019-07-24:
I will keep this issue open.

wwang0, 2019-08-04:
BTW, @aswaterman when vl=0, will vd[0] be updated?

aswaterman, 2019-08-04:
I think these should follow the usual rule that no destination elements are updated when vstart >= vl (i.e. when vl=0, since vstart must be 0).

commit 926649b2 (2019-09-03): Clarified reduction behavior when all vector operands are masked.
Closes #259.
closed 2019-09-03 by kasanovic