riscvarchive/riscv-v-spec #261: about slidedown instruction

wwang0, 2019-07-24:
In slidedown session, it says:
vslidedown behavior for source elements for element i in slide
0 <= i+OFFSET < VLMAX Read vs2[i+offset]
VLMAX <= i+OFFSET Read as 0

shouldn't it be "0 <= i+OFFSET <**vl**, Read vs2[i+offset]“ ?

aswaterman, 2019-07-24:
No, because the source elements may have indices >= vl. vl affects how many
destination elements are written.

wwang0, 2019-07-25:
if the values of slidedown's source register was generated from  an instruction with the same vl  before running slidedown,  the elements between vl and VLMAX are actually zero, so there is no much difference. And if vl is not the same, current definition is more simple nd flexible. OK, I got it. Thanks.

closed 2019-07-25 by wwang0