riscvarchive/riscv-v-spec #262: clarify vid.v about the illegal instruction exception

lihanglin, 2019-07-25:
There is a statement in chapter 16.9 about vid.v instruction

>An  illegal  instruction  exception  is  generated  if  the  destination  vector  register  group  overlaps  the  mask register and LMUL > 1.

Can it be modified as the following sentence to be more clear?

An  illegal  instruction  exception  is  generated  if  the  destination  vector  register  group  overlaps  the mask register **if masked** and LMUL > 1.

aswaterman, 2019-07-25:
Yes, thanks, this looks like a bug in the spec.

aswaterman, 2019-07-25:
Are you sure you aren't looking at an old version of the spec?  It looks like that sentence was deleted a few weeks ago (as it should have been).

closed 2019-07-25 by aswaterman