riscvarchive/riscv-v-spec #263: When vl <=OFFSET < VLMAX, does tail element need to be set 0 for vslideup?

wwang0, 2019-07-25:
As the title. No valid element is changed, do we still need to zero the tail elements?

And can I assume if No valid element(before vl) is changed, tail elements don't need to be zero'd ? an extreme case is vm masks all(no active elements), do we need zero tail elements per our current policy. Thanks

aswaterman, 2019-07-25:
No. Tail zeroing depends only on vl and vstart. It does not depend on the
mask settings (as that would significantly complicate some implementations).

wwang0, 2019-07-25:
OK. How about my first question:
"
When vl <=OFFSET < VLMAX, does tail element need to be set 0 for vslideup?
"
Thanks

wwang0, 2019-07-25:
When vl <= OFFSET < VLMAX, and vl > vstart, we need zero the tail element, and then you will find only tail elements are updated. Is that desired?

aswaterman, 2019-07-25:
When vstart < vl, elements past vl are always zeroed.  These instructions are not unique in that respect.

wwang0, 2019-07-26:
Thanks. So close it.

closed 2019-07-26 by wwang0