riscvarchive/riscv-v-spec #264: Change behavior of vsetvl{i} rd, x0, {imm,rs2} to use vl instead of ~0 as AVL

kasanovic, 2019-07-25:
Currently, if rs1 of vsetvl{i} is x0, vl is set to VLMAX.  The proposal is to change behavior to have rs1=x0 indicate that current vl should be used as AVL.  In the case that SEW/LMUL is unchanged, this value of vl should also be unchanged.   This avoids having to keep AVL in a scalar register.

Obtaining VLMAX will now require a "li t1, ~0; vsetvli t0, t1, ex,my" sequence, but this is less frequently needed than switching SEW/LMUL with vl unchanged.

solomatnikov, 2019-07-25:
What is supposed to happen if SEW is larger and therefore vl should become smaller? AVL, which was equal to previous vl, value lost?

kasanovic, 2019-07-25:
Yes.  We want to preserve invariant that vl cannot be set to illegal value relative to current SEW and LMUL, so vl is fed in as AVL argument.  The normal case is for constant SEW/LMUL where vl doesn't change.

solomatnikov, 2019-07-25:
My point is: how would you strip mining after that?

AVL is fed into vsetvli, which produces new vl < AVL and AVL value is gone because you said AVL is not present in any other register.

aswaterman, 2019-07-25:
I dispute the premise of this issue.  Holding AVL in a register is sufficient but not necessary: you can instead use the current VL as the argument to future setvl instructions that keep SEW/LMUL constant; the new VL will come out the same as if you had used AVL.

In most circumstances, at least one of VL or AVL needs to be held in a register for some other purpose, so at least one of the two is likely to be alive in a scalar register anyway.  This proposal will only help when the scalar registers holding AVL and VL would otherwise both be dead, and there is sufficient register pressure for that to matter.

OTOH, making this change is guaranteed to add one instruction to most first-fault use cases.  So without quantitative justification to the contrary, this proposal seems like a performance regression.

aswaterman, 2019-07-26:
I've been informed that a stronger argument for this change is to better support an intrinsics programming model, where changing vtype without changing vl will be common.  In the general case, the compiler doesn't know the value of AVL/VL to pass into vsetvl, so it would either need to read the CSR to get the current VL, or require the user to pass in the AVL or current VL as another argument to the intrinsic.

I'll drop my objection.

rkruppe, 2019-07-27:
Hold on. It's to early to say definitely about what C-level intrinsics will look like, but "require the user to pass in the AVL or current VL as another argument to the intrinsic" seems quite plausible. It's

- preferable for compilers that want to treat VL as an SSA value rather than a global mutable variable (greatly simplifies analyses and optimizations of vector code *and surrounding code*), and
- perfectly workable for compiler that don't try to be clever (if it's defined to be UB for the VL passed to an operation to not be the result of the last setvl that was executed before, you can just ignore the extra argument)

However, there are also other places where the proposal makes it easier to emit efficient code sequences for vtype changes, e.g., late in the backend where much context may be lost -- either AVL or VL is probably available, but it's difficult for some part of the compiler to figure that out. So I don't necessarily want to pick up the banner of objecting. Still, I want to register that the intrinsics angle doesn't really convince me.

aswaterman, 2019-07-27:
Thanks for that perspective, Robin.

commit ddfadd1f (2019-08-29): Flesh out new vsetvl{i} proposal

Closes #264
commit af3ad34e (2019-08-29): Flesh out new vsetvl{i} proposal

Closes #264
commit 2be4d10d (2019-08-29): Flesh out new vsetvl{i} proposal

Closes #264
commit de7ddeda (2019-08-29): Flesh out new vsetvl{i} proposal

Closes #264
closed 2019-09-09 by kasanovic