riscvarchive/riscv-v-spec #265: Can the destination register of widening reduction instructions overlap with vs2/vs1 source operand?

lihanglin, 2019-07-25:
In chapter 15 the spec. says

>The scalar input and output operands are held in element 0 of a single vector register, not a
vector  register  group,  so  any  vector  register  can  be  the  scalar  source  or  destination  of  a  vector  reduction regardless of LMUL setting.

and the spec. also says in chapter 11.2 about widening instructions

>The destination "vector register group" cannot overlap a source "vector register group" of a different element width (including the mask register if masked), otherwise an illegal instruction exception is raised.

As my understanding the destination and vs1 operand of reduction instructions are not vector register group. and the following instruction sequence are legal without any exception

```asm
vsetvli t0, s8, e16
vwredsumu.vs v0, v0, v0, v0.t
```

correct ?

aswaterman, 2019-07-26:
Since reductions can't produce their result until they've consumed all their operands, there is no need for an overlap constraint.  And I agree with your interpretation that the spec already conveys that, because the destination is not a register group.

yichanfeng, 2019-07-26:
When EDIV = 1, the vd is performed as scalar operand (**vd[0]**, section 15.1).
`vredmax.vs  vd, vs2, vs1, vm   # vd[0] =  max( vs1[0] , vs2[*] )`

When EDIV > 1, it seems that type of **vd** is changed from **vd[0]** to **vd[i]** (section 19.2.3).

```
# Find maximum among sub-elements (_v6_ is **vs1**)
vredmax.vs v5, v6, v7 # **v5[i]**[7:0] = max(v6[i][31:24], v6[i][23:16],v6[i][15:8], v6[i][7:0], v7[i][7:0])
```
Does it means that **vd** is changed from **scalar operand** to **vector register group** ?
If the answer is "NO" => there is no need for an overlap constraint regardless of EDIV.
If the answer is "YES" => there is need for an overlap constraint when EDIV > 1.

Is the **above assumption** correct ?

Thanks!

aswaterman, 2019-07-27:
It looks like reductions under EDIV can produce more than VLEN bits of output, so as currently specified, I believe you are right.

EDIV is currently underspecified and likely to change in significant ways, so perhaps this, too, will change.

closed 2019-08-01 by lihanglin