riscvarchive/riscv-v-spec #266: Overlapping vector register groups in vfwcvt instructions

poinu, 2019-07-25:
Section *11.2. Widening Vector Arithmetic Instructions* states that:

> The destination vector register group cannot overlap a source vector register group of a different element width (including the mask register if masked), otherwise an illegal instruction exception is raised.

I was wondering if this restriction applies to *Widening Floating-Point/Integer Type-Convert Instructions* too. That is, are the following codes valid?

```
vsetvli x0, x0, e32, m1
vfwcvt.x.f.v v2, v2                                                                   
```
```
vsetvli x0, x0, e32, m1
vfwcvt.x.f.v v2, v3                                                                   
```

Thanks!

aswaterman, 2019-07-25:
The floating-point instructions should have the same constraints as the
integer ones.

yichanfeng, 2019-07-26:
Hi, I have the same question. 
Does it means that HW still needs to check the overlapping **even if** destination vector register is **not** a vector register group (e.g., syntax is a vector register. But, it is performed as scalar operand). 
Thanks!

poinu, 2019-07-26:
Thank you Andrew for the answer.

I understand that floating-point widening/narrowing instructions have the same constraints as the integer ones, what isn't clear for me is whether those constraints would also apply to conversions.

If this is the case, perhaps it could be useful to state that such restrictions apply to all widening/narrowing instructions (unless the contrary is specified).

Currently the constraint is explicitly specified for:

- Widening/narrowing arithmetic instructions (11.2/11.3)
- `vslideup`/`vslidedown` instructions (17.4.1/17.4.2)
- The `vrgather` instruction (17.5)
- The `vcompress` instruction (17.6)

aswaterman, 2019-07-26:
I see.  Yes, the conversion operations are constrained similarly.  I will clarify that with a reference back to 11.2/11.3.

aswaterman, 2019-07-26:
Closed by 26581142f9ab4ddfe26f3a1debc59d90e83f0d8f

aswaterman, 2019-07-26:
@yichanfeng since widening reductions can't produce their result until they have read all of their operands, there is no need for an overlap constraint.  For the reason you mentioned (it's not a register group), I don't think this needs any change to the spec.

closed 2019-07-26 by aswaterman

poinu, 2019-07-30:
Thanks for the clarification!