riscvarchive/riscv-v-spec #267: vm mask on vslide* instructions

wwang0, 2019-07-26:
Does the vm mask work against the source register or the destination register in vslide* instructions?

aswaterman, 2019-07-26:
The same rule applies as for other masked instruction: it affects which
destination elements are written.

wwang0, 2019-07-26:
Then I think  the constraints on all vslide* overlapping the mask register can be relaxed: dest register can overlap with mask register when LMUL = 1.

aswaterman, 2019-07-26:
I don't think that would work for vslideup or vslide1up, for the same reason overlap is disallowed on the vs2 argument: you'd destroy part of the mask before you got to use it.

We already have relaxed this constraint for vslidedown (3f117c435d9fc07a6ca43c3d08abb75a1f43cf1f) but it looks like we forgot to make the same change for vslide1down.  I will fix that.

wwang0, 2019-07-28:
@aswaterman ， but mask is just against destimation elements, how can it be destroyed for vslideup/vslide1up?

aswaterman, 2019-07-28:
Hmm, you're right.  I think that overlapping the mask register could be allowed when LMUL=1.  I'll make the pull request.

wwang0, 2019-07-29:
Great!

commit 93a77f54 (2019-07-26): Relax mask register overlap constraint for vslide1down

This follows the same logic as 3f117c435d9fc07a6ca43c3d08abb75a1f43cf1f

Closes #267
PR #271 (2019-09-09): vslideup and vslide1up can overlap the mask register when LMUL=1
Output element i is masked by mask element i, so there are no WAR hazards on the mask register when LMUL=1.

(The Vector Masking section states the weaker LMUL > 1 constraint, so this PR only deletes text.)

Closes #267
commit b853d77d (2019-09-09): vslideup and vslide1up can overlap the mask register when LMUL=1

Output element i is masked by mask element i, so there are no WAR
hazards on the mask register when LMUL=1.

(The Vector Masking section states the weaker LMUL > 1 constraint,
so this PR only deletes text.)

Closes #267
closed 2019-09-09 by kasanovic