riscvarchive/riscv-v-spec #273: For the detection of unsupported IEEE floating-point type, which type of SEW should be used ?

yichanfeng, 2019-07-31:
Hi, I have one question about illegal instruction detection of unsupported IEEE floating-point type.

In chapter 15 (**Vector Floating-Point Instructions**) the spec. says

`The standard vector floating-point instructions treat 16-bit, 32-bit, 64-bit, and 128-bit elements as IEEE-754/2008-compatible values. If the current SEW does not correspond to a supported IEEE floating-point type, an illegal instruction exception is raised.`

Is **current SEW** equal to **effective SEW** ?
If the answer is "NO"  => HW always uses **SEW** to check if floating-point type is supported.
If the answer is "YES" =>  HW uses **SEW** or **2*SEW** to check if floating-point type is supported (e.g., SEW for non-widening/narrowing, 2*SEW for widening).

aswaterman, 2019-08-06:
I think the spec should say that all datatypes involved in the operation must be supported.  For example, when widening from floatX to floatY, the operation is only allowed if both floatX and floatY are supported.

This might be asymmetric between int and float.  For example, on systems that implement float32 and float64 but only implement 32-bit integers, widening from float32 to float64 might be supported, but widening from float32 to int64 might be illegal.

kasanovic, 2019-12-07:
Text of spec needs improving.

commit dc030f23 (2020-05-05): Clarified that all vector FP operand widths used by an instruction must
be supported else an illegal instruction is raised.
Closes #273.
closed 2020-05-05 by kasanovic