riscvarchive/riscv-v-spec #274: Clarify vector-immediate shift instruction behavior

larryrong, 2019-08-01:
The shift amount is described in the following chapters
- 12.5. Vector Single-Width Bit Shift Instructions
Only the low lg2(SEW) bits are read to obtain the shift amount.
The immediate is treated as an unsigned shift amount, with a maximum shift amount of 31.
- 12.6. Vector Narrowing Integer Right Shift Instructions
The shift amount can come from a vector or a scalar x register or a 5-bit immediate. The low lg2(2*SEW) bits of the vector or scalar shift amount value are used (e.g., the low 6 bits for a SEW=64-bit to SEW=32-bit narrowing operation). The unsigned immediate form supports shift amounts up to 31 only.
- 13.5. Vector Single-Width Scaling Shift Instructions
The low lg2(SEW) bits of the vector or scalar shift amount value are used. The immediate form supports shift amounts up to 31 only
- 13.6. Vector Narrowing Fixed-Point Clip Instructions
The low lg2(2*SEW) bits of the vector or scalar shift amount value are used (e.g., the low 6 bits for a SEW=64-bit to SEW=32-bit narrowing operation). The immediate form supports shift amounts up to 31 only.

The shift amount has several explanations in the above chapters. For vector shift instructions, Does vector-immediate shift instructions follow the constraint that is "{lg2(SEW)/lg2(2*SEW)} bits are used"? 
```
e.g.
SEW = 8, lg2(SEW) = 3
A.
    vsra.vi   v16, v8, 26 # 26(0x1a)   v2[i] = v1[i] >> 2
    vnsra.vi  v16, v8, 26 # 26(0x1a)   v2[i] = v1[i] >> 10 (0xa)
    vssra.vi  v16, v8, 26 # 26(0x1a)   v2[i] = v1[i] >> 2
    vnclip.vi v16, v8, 26 # 26(0x1a)   v2[i] = v1[i] >> 10 (0xa)
B.
    vsra.vi   v16, v8, 26 # v2[i] = v1[i] >> 26(0x1a)
    vnsra.vi  v16, v8, 26 # v2[i] = v1[i] >> 26(0x1a)
    vssra.vi  v16, v8, 26 # v2[i] = v1[i] >> 26(0x1a)
    vnclip.vi v16, v8, 26 # v2[i] = v1[i] >> 26(0x1a)

Which one is correct?
```

aswaterman, 2019-08-01:
It's option A.  I don't find the current spec to be ambiguous in this respect.  Remarks like "only the low lg2(SEW) bits" apply to immediate arguments just the same as they apply to dynamic arguments.

larryrong, 2019-08-06:
Clear 
Thank you all.

closed 2019-08-06 by larryrong