riscvarchive/riscv-v-spec #276: What is the most efficient way to copy a scalar register value to one element of register?

wwang0, 2019-08-04:
The psedo operation is like below
vext.v.x vd, rs1, rs2 # vd[rs2] = rs1

In the past we had vext.x.v, to do rd = vs2[rs1], and now it is gone and can be replaced by vslidedownvx followed by vmv.x.s.  But what is efficient way to achieve vext.v.x ? we must use mask register for sure, but the process of changing specified mask element is the same as vext.v.x essentially...

aswaterman, 2019-08-04:
There might be a better way, but here is one approach:

    vid.v v16
    vmseq.vx v0, v16, t0
    vmv.v.x v24, t1, v0.t

This is not terrible, given that the mask operation is fast relative to high-LMUL vector operations.

wwang0, 2019-08-04:
Understood. But how to define 'common' is really a problem ;) Usually how do you guys gather such information to justify an instruction?

aswaterman, 2019-08-04:
Yeah, I know what you mean; in this case it is just intuition.  Given that the application vector length rarely matches the hardware vector length, insertion into the middle of a hardware vector is not likely to be useful too often.  By contrast, insertion at the head or the tail (vslide1up/vslide1down) is more often useful even when the application vector length doesn't match the hardware vector length.

wwang0, 2019-08-05:
I see. Thanks Andrew!

closed 2019-08-05 by wwang0