riscvarchive/riscv-v-spec #277: Fixed point instruction examples needed

JamesKenneyImperas, 2019-08-08:
The description of many of the fixed point instructions in the specification is not sufficiently detailed to make the behavior clear. These instructions are complex because they typically do multiple things (e.g. a widening or narrowing operation, a shift, a round using vxrm).

The required effect should be made explicit by concrete examples or more comprehensive pseudocode that incorporates a precise description of the rounding algorithm and the result bits on which it operates. This is particularly true of vsmul and vwsmaccu etc.

JamesKenneyImperas, 2019-08-16:
As an example of why I think this is necessary, the supplied pseudocode for _vssra.vi_ does not work for some input data values if taken at face value. Suppose that SEW=8, VS2=0x71 and src2=5, with rounding mode RNU. According to section 13.5:

 _# For vxrm=rnu, round = 1 << (src2-1), where src2 is vs1[i], x[rs1], uimm_

In this case, this means round=0x10 (which is 1<<4). The pseudocode then says that the result is:

_# vd[i] = ((vs2[i] + round)>>uimm)_

which is (0x71+0x10)>>5, i.e. 0x81>>5, which is 0xfc (assuming a signed shift of an SEW intermediate). This is not correct: the result should be 0x04. The problem is that the given pseudocode will work _only if the input vs2[i] is implicitly extended to wider than SEW bits before adding round_, and this is not stated, unlike for some other instructions in this section.

This is just one example, of one of the simpler instructions.

JamesKenneyImperas, 2019-09-09:
To elucidate further, I think the fundamental problem with these descriptions is that they are attempting to describe what is necessarily a _discrete sequence of operations_ as an _expression_. I think all descriptions should be rewritten in a functional form. I'm making up syntax here, but it should be something like this:

    int<bits> vssra<bits>(int<bits> vs2, int shift, rounding_mode rm) {

        int<bits> wholePart = vs2 >> shift;
        uns<bits> fraction  = vs2 << (bits-shift);    // normalized fraction (shifted-out bits)

        return wholePart + fixedPointRound<bits>(wholePart, fraction, rm);
    }

Function fixedPointRound takes the whole part, normalized fraction, rounding mode, and returns either 0 or 1. When rewritten like this, no widening is required.

kasanovic, 2019-11-10:
Fixed in db4c604.

closed 2019-11-10 by kasanovic