riscvarchive/riscv-v-spec #278: vfncvt.f.f.v and multiplerounding events

JamesKenneyImperas, 2019-08-09:
There is a statement in section 14.17 which is doubly-misleading. It says:

_A full set of floating-point widening conversions are not supported as single instructions. Conversions can be implemented in several halving steps, with equivalently rounded results and with the same exception flags raised (possibly raised redundantly in multiple steps)._

The first sentence should say "narrowing", not "widening", but even with this change the statement is incorrect, at least for rounding mode RNE. Consider the following double-precision value (split into sign, exponent and fraction fields):
    0.01111111111.1111111111011111111111110000000000000000000000000000
Rounding directly to half-precision using RNE in one step produces:
    0.01111.1111111111 (a)
Going via single precision using RNE in two steps produces:
    0.01111111.11111111111000000000000
followed by:
    0.10000.0000000000 (b)
Values (a) and (b) are not the same.

If the current rounding mode is RNE, then the correct procedure is to perform all but the last step using ROD mode (jamming) and use RNE only for the final step. Since the rounding instructions only use the current rounding mode (unlike the base architecture) this will require two fcsr changes in addition to the rounding operations themselves, which is cumbersome.

Perhaps the vfncvt.f.f.v instruction needs changing to support an explicit rounding mode, or a new instruction added with ROD semantics, or perhaps new instructions added for direct conversions between types.

jhauser-us, 2019-08-09:
JamesKenneyImperas is correct on all counts.

To restate his suggestions, choose one of:
 - Add the full set of narrowing instructions from any size to any other narrower size.
 - Add general support for floating-point "round-to-odd" (not a bad idea), plus modify the half-narrowing conversions to include a rounding mode field.
 - Add a parallel set of half-narrowing instructions with the rounding mode fixed to "round to odd".

aswaterman, 2019-08-09:
Thanks, guys.  Adding round-to-odd seems to fit the existing vector ISA template better than adding the full set of narrowing conversions.

aswaterman, 2019-08-09:
Adding a variant of `vfncvt.f.f.v` that ignores the rounding mode and rounds to odd might be less disruptive than adding the new dynamic rounding mode, since it only affects one operation.  And as James points out, it would save an instruction.

kasanovic, 2019-10-04:
Task group agreed to add the narrowing conversion with jamming.

closed 2019-10-04 by kasanovic