riscvarchive/riscv-v-spec #279: Vector Extension and Privileged State

JamesKenneyImperas, 2019-08-12:
The current Vector Extension specification has "interaction with privileged architectures" as a known issue. I think this is particularly true for floating point state (misa.[FD] and mstatus.FS). This really needs some further clarification, and I therefore have some questions about the intention of the specification - can you please comment?

1. The base Privileged Architecture specification says, in reference to the mstatus.FS field:
_The FS field encodes the status of the floating-point unit, including the CSR fcsr and floating-point data registers f0–f31._
It further states that:
_When an extension’s status is set to Off, any instruction that attempts to read or write the corresponding state will cause an illegal instruction exception._
Since vector floating point instructions read and write the fcsr, is it therefore correct to assume that executing such instructions requires mstatus.FS!=0? If not, what about Vector Extension instructions that read and write f register values?

2. In section 14 of the Vector Extension specification, it states that:
_Platforms supporting 16-bit half-precision floating-point values will also have to implement scalar half-precision floating-point support in the f registers._
2.1. Is the implication of this statement that _if an N-bit floating-point type is supported in the Vector Extension then the same type must also be supported in the f registers_? If not, why are 16-bit half-precision floating-point values special in this respect?
2.2. Assuming that the implication about N-bit floating-point type support is true, can you clarify the meaning of the word _supported_ in this context? For example, if single-precision floating point is supported in the processor (i.e. misa.F=1 at reset) but currently disabled (misa.F is writable and zero currently), would this be expected to cause Vector Extension instructions operating on single-precision data to be illegal or not?
2.3 As an aside, the Unprivileged ISA specification appears currently to have no mention of half-precision support except for one line in a table in section 13.2. Is this documented elsewhere? Required information would be the misa enable bit details and opcode allocations for conversion instructions, for example.

aswaterman, 2019-08-29:
Re: (1): To paraphrase Rumsfeld, this issue is a known unknown, and has also been discussed on this PR: https://github.com/riscv/riscv-v-spec/pull/135 -- we have not reached consensus but are aware we need to do something.

Re: (2): I suggest creating one ticket per issue to reduce the odds of a digressive discussion.

kasanovic, 2019-11-24:
(1) clarified in a6f94e7
Please re-enter 2 as separate issue.

closed 2019-11-24 by kasanovic