riscvarchive/riscv-v-spec #280: vfmerge.vfm/vmerge.vxm and vfmv.v.f/vmv.v.x

JamesKenneyImperas, 2019-08-15:
In section 14.13, there is a note about vfmerge.vfm:
    _In Zfinx systems, the instruction is identical to vmerge.vxm_
And in section 14.14, there is another similar note about vfmv.v.f:
    _In Zfinx systems, the instruction is identical to vmv.v.x_

I think neither note is correct. vmerge.vxm and vmv.v.x will execute with any implemented SEW, but I believe for vfmerge.vfm and vfmv.v.f the specification implies the SEW must be a valid _floating point_ SEW (probably 32 and 64, maybe 16 but definitely not 8). If the SEW is not a valid floating point SEW, the instructions should not execute but take an Illegal Instruction exception instead.

aswaterman, 2019-11-29:
This issue is actually tied to #339.  If we check for NaN boxes, we need to be able to convert to the canonical NaN.  We can only do that if the data type is a valid floating-point type, because otherwise, we don't know what pattern to use for the canonical NaN.

aswaterman, 2019-11-29:
I think the best resolution is to remove these NOTE, regardless of how #339 is resolved.  A more detailed V + Zfinx spec will eventually be necessary can delve into these issues.

commit 34c39ec2 (2019-11-29): Remove misleading NOTEs on vfmerge/vfmv

Closes #280
closed 2019-11-29 by aswaterman