riscvarchive/riscv-v-spec #284: When vl=0, will scalar (non-vector) destination register be updated?

chuanhua, 2019-09-04:
In section 3.3, a note states that

> As a consequence, when vl=0, no elements are updated in the destination vector
> register group, regardless of vstart.

This only talks about destination vector registers. What is the correct behavior if the destination register is a scalar (non-vector) register? For example, vmv.x.s, vfmv.f.s, vpopc.m, vfirst.m instructions?

aswaterman, 2019-09-04:
By being silent on the matter, the spec implies that the scalar destination register is always written.  We should add a note to that effect (but there's no need for additional normative text).

This definition simplifies designs with scalar register renaming, and seems to be neutral for designs without.

The value written is unambiguous: vmv.x.s and vfmv.f.s copy element 0 (regardless of vl), and for the mask ops, there are no active elements, so e.g. vfirst.m returns -1 and vpopc.m returns 0.

chuanhua, 2019-09-06:
It would be great if you could add such notes at the locations where each such instruction is being described. Thanks!

PR #288 (2019-09-09): Add note that scalar regs are updated even when vstart >= vl
Resolves #284
commit a7f8c580 (2019-09-09): Add note that scalar regs are updated even when vstart >= vl

Resolves #284
closed 2019-09-09 by kasanovic