riscvarchive/riscv-v-spec #289: widening left shift

jnk0le, 2019-09-08:
Current spec is missing widening left shift instruction from symmetry with narrowing right shifts.
It can be substituted with dummy widening addition followed by shift or widening multiply. But both cases require more instructions/latency and multipliers may be expensive as well.

kasanovic, 2019-10-04:
Instruction added.

kasanovic, 2019-10-04:
And instruction was deleted after task group discussion, after it was felt is was not important enough to justify as single instruction.
Widening multiply can substitute with single vector instruction (plus setting up scale amount), and widen+shift in other cases.  Providing all sign-extension cases would have expanded opcode footprint substantially.

closed 2019-10-04 by kasanovic

aswaterman, 2019-10-04:
Thank you—this would have been an irregularity in the implementation I’m working on.

commit 7946a289 (2019-10-04): Removed widening left shifts.
Closes #289.