riscvarchive/riscv-v-spec #299: inconsistent of averaging add description

HanKuanChen, 2019-10-01:
According to [averaging add](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#132-vector-single-width-averaging-add-and-subtract),
> The averaging add and subtract instructions **right shift the result by one bit** and **round off the result according to the setting in vxrm**.

However, the description in code is
```
# result = (src1 + src2 + round) >> 1;
```

I think it should be
```
# result = ((src1 + src2) >> 1) + round;
```

JamesKenneyImperas, 2019-10-01:
I agree. Also note that the expression:

`    ((src1 + src2) >> 1)`

must be evaluated with operands sign-extended to SEW+1 bits for the result to be correct, and this is not made clear. For example, with SEW=8, an averaging add of -128 and -128 (i.e. 0x80 and 0x80) should produce a result 0x80, not 0x00 (which would be the case if not extended).

See also issue #277.

kasanovic, 2019-11-10:
Rounding description issues fixed in commit db4c604.

kasanovic, 2019-11-10:
#256 discusses unsigned average add/sub instructions so closing this issue here.

closed 2019-11-10 by kasanovic