riscvarchive/riscv-v-spec #300: Floating Compare results for NaN arguments

billhuffman, 2019-10-04:
Section 14.11 on floating compares is clear on flag-setting for all operations.  For results, it refers to the scalar floating-point compare spec.  This answers the question for EQ, LT, and LE, but doesn't quite answer for NE, GE, and GT.

NE might be zero if either input is a NaN or it might be the NOT of EQ.

GE/GT are probably zero if either input is a NaN so that they are the "scalar-vector" or (reverse order) version of LT/LE - but it's not clear.

chuanhua, 2019-11-29:
According to IEEE754-2008 standard Table 5.1, NOT(=) predicate is a negation of EQ predicate. And this predicate negation result is the same as the result shown in Table 4 of the IEEE754-1985 standard.

So vmfne with NaN as a source operand should return a result of 1 instead of 0. And it is the same as !vmfeq.

We agree with Bill that this needs to be clarified in the vector spec.

aswaterman, 2019-11-29:
Yes, NE should be the negation of EQ, whereas e.g. LE is not the negation
of GT. This deserves a note in the spec.

chuanhua, 2019-11-29:
Great! Thanks.

aswaterman, 2019-11-29:
Closed by 2658b8c85ee4cce0085e1c87d3f61646baa0d26a; please sanity-check my description.

billhuffman, 2019-12-02:
Looks good to me, Andrew.

      Bill

On 11/29/19 1:03 PM, Andrew Waterman wrote:

Closed by 2658b8c<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_riscv_riscv-2Dv-2Dspec_commit_2658b8c85ee4cce0085e1c87d3f61646baa0d26a&d=DwMCaQ&c=aUq983L2pue2FqKFoP6PGHMJQyoJ7kl3s3GZ-_haXqY&r=AYJ4kbebphYpRw2lYDUDCk5w5Qa3-DR3bQnFjLVmM80&m=elhcPy0O7How7YIB9Hhlp-ToKj6kdOq49kEpqhOC6Iw&s=RIjKX-8rppFmsQO2Q8lXW9DF6rO0epiKAl9Mzm4TmbA&e=>; please sanity-check my description.

closed 2019-12-07 by kasanovic