riscvarchive/riscv-v-spec #301: There are two different syntaxes for mask register.

JerryShih, 2019-10-05:
@aswaterman 
Why we have two different usage for mask?
Here are the example:
1) vfsgnjn.vf vd, vs2, rs1, **v0.t**
2) vmerge.vxm vd, vs2, rs1, **v0**

rkruppe, 2019-10-05:
IIRC the rationale in the meeting where this was decided was:

- `v0.t` syntax should be reserved for when the mask controls writeback, i.e., destination elements are undisturbed where the mask has 0s
- For `vmerge`, the mask register is not used in that way, instead all destination elements (up to vl) are written unconditionally (just the elements written depend on the mask bits)

rkruppe, 2019-10-05:
See also discussion in https://github.com/riscv/riscv-v-spec/issues/189

aswaterman, 2019-10-05:
Sounds like a good explanation to me.

JerryShih, 2019-10-09:
@rkruppe 
Thank you.

closed 2019-10-09 by JerryShih