riscvarchive/riscv-v-spec #302: Is `ELEN` should be known in compile time?

zakk0610, 2019-10-10:
According to Krste's [slide]( https://content.riscv.org/wp-content/uploads/2019/06/17.40-Vector_RISCV-20190611-Vectors.pdf#page=6 ), it said `Vector ISA designed to allow same binary code to work across variations in VLEN and SLEN`, so is it mean the toolchain (or others) need to take care about not portable part of implementation-defined parameters?
For example, in widening instruction case, if destination `2*SEW` is larger than `ELEN`, user will get the illegal instruction exception in runtime.
But if `ELEN` is known in compile time, maybe compiler can check the widening instruction is legal or not for given `ELEN` value, or maybe loader can do a check the elf's ELEN is workable on current HW config.

aswaterman, 2019-10-10:
The compiler should be able to assume (or be told by the user) the minimal value of ELEN. ELEN of course might be larger on the implementation.

closed 2019-10-11 by zakk0610

ebahapo, 2020-06-18:
Since ELEN≥`max(XLEN, FLEN)`, perhaps it suffices to record the ABI for compatibility.  Specifically, ELEN≥32 for rv32f and ELEN≥64 for rv{32,64}d and rv64f.